meta-rust/meta-rust #1: Native linker not found when building libgit2_git

posborne, 2015-03-10:
I haven't tried forcing to the versioned libgit2 (by default bitbake picks the git version).  Here's the failing part of the build for me using the latest meta-rust:

```
CMAKE_AR-NOTFOUND cr libgit2.a  CMakeFiles/git2.dir/src/pack-objects.c.o ...
Error running link command: No such file or directory
```

jmesmon, 2015-03-10:
Do you have poky commit 60d504b431fa7cdc6ded141def7ae18edb1834e5 or oe-core commit 79144da00f005b5a3ab8f7404730216cfc684616 ?
```
cmake: supply CMAKE_AR in toolchain file
```

It might address your issue, though I can't recall the reason I submitted it in the first place, it may well have been to avoid this error.

jmesmon, 2015-03-10:
I've just noticed that I set PREFERRED_VERSION to the fixed version one locally, and the srcrev for the -git version is fairly out of date. I've removed the -git version, let me know if that fixes it for you.

posborne, 2015-03-11:
I'm still seeing the issue when using the recipe tor 0.22.1.  Here's the layers I have setup right now:

```
$ bitbake-layers show-layers
layer                 path                                      priority
==========================================================================
meta                  /home/posborne/Projects/rust-playground/rustbone/build/../poky/meta  5
meta-yocto            /home/posborne/Projects/rust-playground/rustbone/build/../poky/meta-yocto  5
meta-yocto-bsp        /home/posborne/Projects/rust-playground/rustbone/build/../poky/meta-yocto-bsp  5
meta-oe               /home/posborne/Projects/rust-playground/rustbone/build/../meta-openembedded/meta-oe  6
meta-rust             /home/posborne/Projects/rust-playground/rustbone/build/../meta-rust  7
meta-rustbone         /home/posborne/Projects/rust-playground/rustbone/build/../meta-rustbone  8
```

Where yocto and meta-openembedded repos are on the latest dizzy.

I'll do some digging to see if I can figure out the root cause and submit a PR.  I should have another PR coming on an issue I ran into with the rust recipe.

jmesmon, 2015-03-11:
Yep, it doesn't look like dizzy has that change (I'm using the latest yocto master). I'm not sure there is a sane workaround here.

Perhaps:

libgit2.inc
```
OECMAKE_AR ?= "${AR}"
cmake_do_generate_toolchain_file_append() {
	cat >> ${WORKDIR}/toolchain.cmake <<EOF
set( CMAKE_AR ${OECMAKE_AR} CACHE FILEPATH "Archiver" )
EOF
}
```

No idea if that will actually manage to do the trick.
Alternately you could append it to another function that happens to be called, or add_task it.
I don't have the bandwidth to support dizzy, but If someone puts together a PR that fixes this for dizzy and avoids breaking master, I'd be happy to merge it.

posborne, 2015-03-11:
The change you suggested does appear to work.  I *think* that it should not cause any regressions upstream either as CMake should have no problem with CMAKE_AR being set multiple times.

I guess the standard way of handling this might be to have a dizzy branch in the meta-rust layer, but if master can reasonably support master and dizzy, that seems easier to me.  I'm willing to test and do some maintenance work to ensure dizzy remains functional.

closed 2015-03-17 by jmesmon