meta-rust/meta-rust #12: Consider having the entire crates.io index/db automatically imported as bb packages

jmesmon, 2015-07-31:
This would also include adding a crates.io fetcher to get the source code for a crate.

posborne, 2016-05-01:
What are your ideas on how this would be implemented?  Code generation that generates actual bb recipe files for each crate or something that doesn't require text files to be written out?

jmesmon, 2016-05-02:
Ideally we wouldn't have the actual text files, but instead add the idea of a recipe into bitbake using the crates.io index git repo as the source for data. Unclear if that is possible, though (I haven't dug into the python hooks in bitbake extensively enough yet).

Which is why I'd see auto-generation of the text files as a first step. At the point where we do have auto-generation though, it would probably make sense to have a separate layer which was automatically updated from crates.io's changes rather than pouring a bunch of automatic changes into `meta-rust`.

PR #58 (2016-07-14): add a custom fetcher for Cargo crates
This adds a custom fetcher for Cargo crates and hopefully will simplify adding Cargo based projects to meta-rust. The URL syntax is:

`crate://HOST/NAME/VERSION` with a standard example being `crate://crates.io/memchr/0.1.11`

Additionally this supports fetching down a snapshot of the registry index for reproducible builds with the syntax:

`crate-index://HOST/REVISION` with an example being `crate-index://crates.io/3b3994e099281c394a6a66604d1af6c0920e4c31`

`crates.io` as a HOST magically fixes up the paths to hit the official registry. This should be abl [... cut]
closed 2016-07-14 by derekstraka