meta-rust/meta-rust #17: use bitbake installed jemalloc

jmesmon, 2015-08-26:


cardoe, 2016-07-12:
I haven't been able to do this on Gentoo due to the patching that happens to jemalloc in the Rust build process. It alters the API and ABI of jemalloc.

cardoe, 2016-11-22:
I looked again at this when bumping Rust in Gentoo and there are further changes from stock jemalloc's API and ABI which IMHO render the two as significantly different enough to not make this worth it. @jmesmon Your thoughts?

jmesmon, 2016-11-22:
This ticket was primarily a reminder to take a look at using non-integrated jemalloc at some point.

As it sounds like (from your investigation) that that isn't (easily) possible, it should be fine to close this.

cardoe, 2016-11-22:
Alright. I'll periodically look at this and should the deltas close a little bit we can re-open.

cardoe, 2016-12-04:
They're breaking jemalloc out into its own crate, so this should resolve itself.

https://github.com/rust-lang/rust/issues/36963
https://crates.io/crates/jemallocator

closed 2016-12-04 by cardoe