meta-rust/meta-rust #23: build system is assumed to be linux-x86_64

jmesmon, 2015-12-13:
For rust & cargo we download snapshots to bootstrap. Right now, these are hardcoded to the linux-x86_64 versions. rust & cargo provide a few other architectures (linux-i686 being the most probably useful), so it'd be useful if:

 - We failed early when we can't support the build system
 - We download snapshots based on the build os & build arch.

cardoe, 2016-11-06:
I was always under the impression that Yocto required the build host to be either x86_64 or i686/i586. Are there other architectures past those that Yocto supports?

jmesmon, 2016-11-06:
I'm not aware of any specific restrictions or allowances for the build architectures used in yocto.

cardoe, 2016-11-22:
So this should be possible to do now with a bbappend that overrides the hashes of the stage0 toolchains. @jmesmon Not sure if you want to define some kind of scope on this ticket so we have an idea of what complete is and isn't.

jmesmon, 2016-11-22:
Goal here was primarily to form the recipes so that the data for multiple build archs could be defined at once, so we'd have a path to add support for new ones as people needed it.

I don't think being able (or being required) to override hashes/locations to allow a different build arch is really completion.

On possible solution is to use [overrides](http://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#var-OVERRIDES) to complete this. Something like:

```
OVERRIDES_append = ":build-${BUILD_ARCH}-${BUILD_OS}"

RUSTC_STAGE0_SRCREV_build-x86_64-linux = "..."
RUSTC_STAGE0_SRCREV_build-x86-linux = "..."
RUSTC_STAGE0_SRCREV_build-arm-linux = "..."
```

And similar for other variables. Will likely want to select slightly different value for OVERRIDES, and confirm that there aren't any likely override conflicts.

I held off on doing this as it didn't appear that this approach was used anywhere else in yocto, so I wasn't quite sure it was the right way of handling things.

cardoe, 2016-11-22:
I agree that this isn't complete I just wanted to scope out what complete was. Thanks for the link about overrides. It was something I didn't know about. You and @derekstraka certainly have more Yocto experience than I do and I'm more or less equivalent to a billion monkeys pounding on a keyboard until it builds