meta-rust/meta-rust #26: rust-llvm/1.5.0-r0/build/BuildTools/Release/lib/libLLVMTableGen.a: error adding symbols: Archive has no index; run ranlib to add one | collect2: error: ld returned 1 exit status

kteza1, 2016-01-22:
Hi. I'm facing this error while building rust-llvm

```
rust-llvm/1.5.0-r0/build/BuildTools/Release/lib/libLLVMTableGen.a: error adding symbols: Archive has no index; run ranlib to add one
| collect2: error: ld returned 1 exit status
```

Any Idea?

jmesmon, 2016-01-22:
I don't think I've seen that before. Can you post a longer extract of the error, the command you're invoking to do the build, and the configuration (layers & otherwise) that you're using?

kteza1, 2016-01-23:
Fetched jethro and built it with out meta-rust and build was successful

```
git clone --branch jethro --single-branch git://git.yoctoproject.org/poky
bitbake core-image-minimal 
```

Below is my configuration

```
Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "beaglebone"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0.1"
TUNE_FEATURES     = "arm armv7a vfp neon callconvention-hard cortexa8"
TARGET_FPU        = "vfp-neon"
meta              
[log.do_compile.txt](https://github.com/jmesmon/meta-rust/files/101795/log.do_compile.txt)

meta-yocto        
meta-yocto-bsp    = "jethro:b9ad87b18fd7a664b8d5e385ffb09496c679036c"
meta-rust         = "master:6f9559d152347e1a7911893c8c6dce4259798149"
```

Cloned meta-rust (master) and added to bblayers.conf and

```
bitbake rust-llvm
```

And it fails

kteza1, 2016-01-25:
Same error with master as well

```
Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "beaglebone"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0+snapshot-20160125"
TUNE_FEATURES     = "arm armv7a vfp  neon        callconvention-hard        cortexa8"
TARGET_FPU        = "hard"
meta              
meta-yocto        
meta-yocto-bsp    = "master:c192bd60e8c7bcb9bbdd24158bd66eb691bfee5a"
```

kteza1, 2016-02-01:
This is occurring only when I do `bitbake rust-llvm` separately. `bitbake rust-hello-world` is working

jmesmon, 2016-02-02:
Huh, that's unexpected. Seems like it might be some lack of dependency in `rust-llvm` that is included via something else.

kteza1, 2016-02-03:
Yeah. Do you think I should close this issue?

jmesmon, 2016-02-03:
I'd leave it: building rust-llvm separately should really work, even if it isn't something people generally need to do. And it's always possible that this issue does show up when building the "normal" way, just not as frequently.

srwalter, 2016-09-01:
I'd actually vote in favor of closing this.  rust-llvm-native is the package that's needed for building rust code, so that's what bitbake is building when you "bitbake rust-hello-world."  rust-llvm would only be needed for building an image containing a compiler, which is separately known not to work and shouldn't be tracked by this issue.

closed 2016-09-01 by srwalter