meta-rust/meta-rust #27: update to rust-1.6.0

jmesmon, 2016-01-22:


kteza1, 2016-02-03:
:+1: Waiting for this

derekstraka, 2016-03-18:
We have a preliminary PR up [here](https://github.com/starlab-io/meta-rust/pull/10) for rust 1.7.0

Once we do a bit more validation, I'll make a PR into here too. (early testing is always appreciated :smile: )

posborne, 2016-03-18:
@derekstraka @cardoe Thanks for putting together this change and the discussions with the rust core team (hopefully in the next few versions we can eliminate the need to patch the rust source completely).  I tested with this branch in combination with [this image](https://github.com/posborne/rustbone) and things build and seem to work properly.

:+1: for the change -- I think it makes sense to open up a PR against this repo sooner rather than later.

PR #44 (2016-03-24): Update rust to version 1.7.0
resolves #27
closed 2016-03-24 by jmesmon