meta-rust/meta-rust #28: build rust std (-runtime) separately from the compiler & eliminate the rustc-cross package

jmesmon, 2016-01-22:
As this avoids building rustc twice, it would also allow merging rustc & rustc-llvm back together without cost.

cardoe, 2016-04-05:
I've been trying to figure out how to do this and honestly their build system really doesn't support it. They're switching to their new build system called rustbuild and I'm hoping they add that support. @brson previously told me on IRC that I'll need to build the whole thing and grab just the parts I want into a package. This is obviously problematic for Gentoo and Yocto since it will cause extra long builds. I've opened a ticket upstream at rust-lang/rust#32597 which will hopefully go somewhere.

brson, 2016-04-05:
It should be much easier to make this work with the new build system but only @alexcrichton knows the details.

alexcrichton, 2016-04-05:
The new rustbuild system may not quite have support for this just yet, but it should in theory be *far* easier to modify it to do so!

jmesmon, 2016-09-25:
This has already been done & merged

closed 2016-09-25 by jmesmon