meta-rust/meta-rust #34: Cargo build error

kteza1, 2016-02-13:
Hi. I'm facing this error while doing `bitbake cargo`. 

```
DEBUG: Executing shell function do_compile
NOTE: make -j 36 -e MAKEFLAGS= ARGS=--verbose
/usr/bin/python src/etc/dl-snapshot.py x86_64-linux
running: curl -o target/dl/cargo-nightly-x86_64-unknown-linux-gnu.tar.gz https://static.rust-lang.org/cargo-dist/2015-04-02/cargo-nightly-x86_64-unknown-linux-gnu.tar.gz
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/etc/bash_completion.d/cargo
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/manifest.in
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/share/zsh/site-functions/_cargo
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/share/doc/cargo/LICENSE-MIT
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/share/doc/cargo/README.md
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/share/doc/cargo/LICENSE-APACHE
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/share/doc/cargo/LICENSE-THIRD-PARTY
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/share/man/man1/cargo.1
extracting cargo-nightly-x86_64-unknown-linux-gnu/cargo/bin/cargo
touch target/snapshot/bin/cargo
/home/ubuntu/data/ather-build/build/tmp/sysroots/x86_64-linux/usr/bin/rustc -V
rustc 1.5.0-dev
target/snapshot/bin/cargo --version
cargo 0.0.1-pre-nightly (84d6d2c 2015-03-31) (built 2015-04-01)
target/snapshot/bin/cargo build --target x86_64-linux --release  --verbose
    Updating registry `https://github.com/rust-lang/crates.io-index`
Failed to parse registry's information for: url

Caused by:
  the given version requirement is not implemented, yet
make: *** [cargo-x86_64-linux] Error 101
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /home/ubuntu/data/ather-build/build/tmp/work/x86_64-linux/cargo-native/0.7.0-r0/temp/log.do_compile.51984)
```

Any work arounds for this?

jmesmon, 2016-02-13:
I had thought I fixed this issue a few commits back, see https://github.com/jmesmon/meta-rust/commit/ec117587bc8d042accaec89de60ff7284f611a8b

Try:
 - make sure you have the above commit
 - `bitbake -c cleanall cargo`
 - `bitbake cargo`

Also note that I don't typically build plain cargo (only cargo-native), so it's a bit less tested. There might be something about building a cargo for target that is making things not quite work right.

kteza1, 2016-02-16:
That's working. Thanks.

closed 2016-02-16 by kteza1