meta-rust/meta-rust #47: build failures due to missing llvm symbols using `cxx11` references

jmesmon, 2016-04-13:
Doesn't happen on all systems. I've observed this on arch linux, but not on gentoo. Might depend on whether the system gcc has the new cxx11 abi enabled.

Probably due to `uninative.bbclass` from selectively disabling gcc's new cxx11 abi:
```
# https://wiki.debian.org/GCC5
# We may see binaries built with gcc5 run or linked into gcc4 environment
# so use the older libstdc++ standard for now until we don't support gcc4
# on the host system.
BUILD_CXXFLAGS_append = " -D_GLIBCXX_USE_CXX11_ABI=0"
```

Full error output
```
| error: linking with `gcc` failed: exit code: 1
| note: "gcc" "-Wl,--enable-new-dtags" "-Wl,--as-needed" "-m64" "-L" "x86_64-linux/stage0/lib/rustlib/x86_64-linux/lib" "x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/bin/rustc.0.o" "-o" "x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/bin/rustc" "-Wl,--gc-sections" "-Wl,-O1" "-nodefaultlibs" "-L" "x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-L" "/home/cody/obj/y/tmp/sysroots/x86_64-linux/usr/lib" "-L" "x86_64-linux/stage0/lib/rustlib/x86_64-linux/lib" "-Wl,-Bstatic" "-Wl,-Bdynamic" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_driver-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_passes-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_borrowck-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_privacy-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_typeck-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_lint-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_resolve-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "syntax_ext-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_plugin-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_metadata-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_trans-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_platform_intrinsics-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_mir-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_data_structures-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_back-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "flate-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "getopts-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "arena-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "fmt_macros-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "graphviz-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_front-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "syntax-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "term-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rustc_llvm-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "rbml-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "serialize-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "log-ca1c970e" "-L" "/home/cody/obj/y/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib" "-l" "std-ca1c970e" "-l" "pthread" "-l" "edit" "-l" "dl" "-l" "m" "-l" "stdc++" "-l" "dl" "-l" "pthread" "-l" "gcc_s" "-l" "c" "-l" "m" "-l" "rt" "-Wl,-rpath,$ORIGIN/../lib" "-Wl,-rpath,/usr/local/lib/rustlib/x86_64-linux/lib" "-l" "compiler-rt" "-L/home/cody/obj/y/tmp/sysroots/x86_64-linux/usr/lib" "-L/home/cody/obj/y/tmp/sysroots/x86_64-linux/lib" "-Wl,-rpath-link,/home/cody/obj/y/tmp/sysroots/x86_64-linux/usr/lib" "-Wl,-rpath-link,/home/cody/obj/y/tmp/sysroots/x86_64-linux/lib" "-Wl,-rpath,/home/cody/obj/y/tmp/sysroots/x86_64-linux/usr/lib" "-Wl,-rpath,/home/cody/obj/y/tmp/sysroots/x86_64-linux/lib" "-Wl,-O1" "-Wl,-rpath=../../lib"
| note: x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib/librustc_llvm-ca1c970e.so: undefined reference to `llvm::TargetRegistry::lookupTarget(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
| x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib/librustc_llvm-ca1c970e.so: undefined reference to `llvm::RTDyldMemoryManager::getSymbolAddressInProcess(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
| x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib/librustc_llvm-ca1c970e.so: undefined reference to `llvm::sys::DynamicLibrary::getPermanentLibrary(char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
| x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib/librustc_llvm-ca1c970e.so: undefined reference to `llvm::createPrintModulePass(llvm::raw_ostream&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
| x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib/librustc_llvm-ca1c970e.so: undefined reference to `llvm::report_fatal_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
| x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib/librustc_llvm-ca1c970e.so: undefined reference to `llvm::SectionMemoryManager::finalizeMemory(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
| x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib/librustc_llvm-ca1c970e.so: undefined reference to `llvm::Triple::normalize[abi:cxx11](llvm::StringRef)'
| x86_64-linux/stage0/lib/arm-poky-linux-gnueabi/rustlib/x86_64-linux/lib/librustc_llvm-ca1c970e.so: undefined reference to `llvm::RTDyldMemoryManager::getPointerToNamedFunction(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
```

hmvp, 2016-07-14:
I am seeing this on Ubuntu 16.04 x86_64 with up to date krogoth build for beaglebone

cardoe, 2016-07-18:
You probably didn't see this on Gentoo because you weren't using gcc-5 there at the time. It's starting to crop up. https://bugs.gentoo.org/show_bug.cgi?id=571600

hmvp, 2016-07-20:
```
diff --git a/src/librustc_back/target/linux_base.rs b/src/librustc_back/target/linux_base.rs
index d1ab71e..fc2ed69 100644
--- a/src/librustc_back/target/linux_base.rs
+++ b/src/librustc_back/target/linux_base.rs
@@ -28,7 +28,7 @@ pub fn opts() -> TargetOptions {
             "-Wl,--as-needed".to_string(),
 
             // Always enable NX protection when it is available
-            "-Wl,-z,noexecstack".to_string(),
+            "-Wl,-z,noexecstack,-D_GLIBCXX_USE_CXX11_ABI=0".to_string(),
         ],
         position_independent_executables: true,
         exe_allocation_crate: super::maybe_jemalloc(),
```
as patch seems to help

jmesmon, 2016-08-08:
I've done this in my yocto config to avoid failures here. This disables all use of the CXX11 ABI. Probably not necessary to be this heavy handed.
```
BUILD_CXXFLAGS_append = " -D_GLIBCXX_USE_CXX11_ABI=0"
TARGET_CXXFLAGS_append = " -D_GLIBCXX_USE_CXX11_ABI=0"
CXXFLAGS_append = " -D_GLIBCXX_USE_CXX11_ABI=0"

BUILD_CXXFLAGS_remove_pn-gcc-runtime = "-D_GLIBCXX_USE_CXX11_ABI=0"
TARGET_CXXFLAGS_remove_pn-gcc-runtime = "-D_GLIBCXX_USE_CXX11_ABI=0"
CXXFLAGS_remove_pn-gcc-runtime = "-D_GLIBCXX_USE_CXX11_ABI=0"
```

Note that without the `_remove` bits gcc-runtime tries to build 2 non-cxx11-abi variants instead of one with cxx11-abi and one without. This duplication shows up as a link failure due to duplicate symbols.

PR #93 (2016-10-02): Ensure proper CXXFLAGS & CFLAGS are passed to rust's `rustllvm` component (c++ files built as part of rustc)
Fixes #47 

Fixes #65  (though might have actually been fixed earlier).

Also includes a few other cleanups.

These changes are extracted from #85 , just removing the commit that merges the building of rust's llvm back into rust itself.
closed 2016-10-02 by cardoe