meta-rust/meta-rust #48: Upgrade to 1.8 stable

realglotzi, 2016-04-19:
There is a new stable version 1.8 on the road and it would be great to have an update of meta-rust.

alexhumphreys, 2016-05-25:
Any news on this issue?

kteza1, 2016-07-11:
Any plans for 1.10 soon?

hmvp, 2016-07-26:
I tried upgrading but between my limited knowledge of the rust build system and issue #47 I could not make it work yet
Here is what I tried: https://github.com/hmvp/meta-rust/tree/try-upgrade-rust

posborne, 2016-07-26:
I am also very interested in moving to a newer version.  @hmvp I tried your branch a few days ago but ran into some other issues when building the rustlib recipe and haven't been able to fully debug as of yet.

cardoe, 2016-08-06:
Honestly your issue is likely related to the change of the target triples. We need to move to the standard ones and away from using the Yocto ones. I made the change in [Gentoo Linux](https://github.com/gentoo/gentoo-rust/blob/master/dev-lang/rust/) with less code and less hassle between upgrades but I haven't been successful with Yocto.

cardoe, 2016-08-06:
I should also mention that I've been working on trying to push changes we have in this layer upstream which means the changes are currently landing for 1.12 so I think we end up targeting 1.10 for a krogoth branch and potentially 1.12 for master. 

/cc @derekstraka

closed 2016-09-01 by srwalter