meta-rust/meta-rust #49: Cargo couldn't fetch a repo

kteza1, 2016-04-29:
Hi. I'm observing this in yocto devshell

```
bitbake -c devshell some_recipe
cargo new test && cd test
```

My `Cargo.toml`

```
[package]
name = "test"
version = "0.1.0"
authors = ["kteza1 <k.teza1@gmail.com>"]

[dependencies.mosquitto]
git = "https://github.com/kteza1/mosquitto-rs"
```

build

```
cargo  build --verbose
    Updating git repository `https://github.com/kteza1/mosquitto-rs`
Unable to update https://github.com/kteza1/mosquitto-rs#40a1202b

Caused by:
  failed to fetch into /home/ubuntu/data/s340-dev-build/build/tmp/work/armv7ahf-vfp-neon-poky-linux-gnueabi/vcu-cloud-connect/1.0-r0/cargo_home/git/db/mosquitto-rs-e9af7a6f8dc35151

Caused by:
  [12/-1] Unsupported URL protocol
```

What might be the reason for this? Any temporary fix ?

posborne, 2016-04-29:
The dependency you have setup will not work on your PC either.  You are missing ".git" on the end of the github repo URL -- you need to copy from the clone URL, not the URL from your web browser (for HTTP, they are the same with .git added).

kteza1, 2016-06-24:
I'm finding this issue only in fido. jethro seems fine. Closing this.

closed 2016-06-24 by kteza1

jmesmon, 2016-06-24:
It's possible this is due to something not picking up the SSL library properly (either curl or libgit2 itself) as well.

kteza1, 2016-06-25:
@jmesmon Not sure. I added all the recipes (from jethro) that I thought were relevant. (https://github.com/kteza1/meta-rust/tree/master/recipes-deps)