meta-rust/meta-rust #53: failure in docker

kteza1, 2016-06-21:
Hi. I'm trying to use meta-rust in a docker container. `bitbake rust-hello-world` is throwing this error

```
x86_64-linux/stage1/bin/arm-poky-linux-gnueabi/rustc: symbol lookup error: /home/build/buildbot/slave-jethro/jeth/main/build/tmp/work/x86_64-linux/rust-cross-a
rm/1.7.0-r0/build/x86_64-linux/stage1/lib/arm-poky-linux-gnueabi/libterm-6a154fe0.so: undefined symbol: _ZN2io5error18Error.error..Error11description20h97638b304e7c4d084tgE

| x86_64-linux/stage1/bin/arm-poky-linux-gnueabi/rustc: symbol lookup error: /home/build/buildbot/slave-jethro/jeth/main/build/tmp/work/x86_64-linux/rust-cross-arm/1.7.0-r0/build/x86_64-linux/stage1/lib/arm-poky-linux-gnueabi/libterm-6a154fe0.so: undefined symbol: _ZN2io5error18Error.error..Error11description20h97638b304e7c4d084tgE
``
Any insights?

cardoe, 2016-06-29:
We use this in a docker container always. Docker isn't the issue here. What's the distro that your container is based on? @derekstraka have we seen something like this? I vaguely remember something like that.

kteza1, 2016-06-29:
@cardoe I tried both ubuntu 14.04 & 15.10 containers. (but it works properly outside container on my ubuntu 15.10 machine)

derekstraka, 2016-06-29:
@cardoe - we did see an error similar to this.  I don't recall if we just removed our SSTATE_DIR and it built or if I added a fix in our fork.  

@RT - Any chance you could try https://github.com/starlab-io/meta-rust/ to see if it builds?  If it does, that will give us an idea of what may be the underlying issue.

kteza1, 2016-06-29:
@derekstraka @cardoe Oh yeah. `libterm` undefined symbol error is gone with `starlab-io/meta-rust`. But unfortunately, I'm facing the below error [which I previously reported](https://github.com/meta-rust/meta-rust/issues/49) (this was a problem in fido but not jethro)

```
cargo: /home/build/buildbot/slave-jeth/jeth/main/build/tmp/sysroots/x86_64-linux/usr/bin/cargo
| rustc: /home/build/buildbot/slave-jeth/jeth/main/build/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/rustc
| NOTE: cargo build -v --target arm-poky-linux-gnueabi --release
|     Updating git repository `https://github.com/kteza1/asystemd.rs`
| Unable to update https://github.com/kteza1/asystemd.rs#6c594f2d
| 
| Caused by:
|   failed to clone into: /home/build/buildbot/slave-jeth/jeth/main/build/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/vcu-cloud-connect/1.0-r0/cargo_home/git/db/asystemd.rs-7b371576c8b86303
| 
| Caused by:
|   [12/-1] Unsupported URL protocol
| WARNING: exit code 101 from a shell command.

```

derekstraka, 2016-06-29:
@kteza1 - Is it safe to assume you are building krogoth or master for your poky version?  Or are you using jethro?

kteza1, 2016-06-29:
@derekstraka I'm on jethro (2.0.2)

derekstraka, 2016-06-29:
@kteza1 - Ok.  I've seen the issue before on krogoth, but we haven't yet moved to use that internally.

kteza1, 2016-06-29:
@derekstraka Ok. Is it openssl? I'm seeing openssl 1.0.2h being built with `starlab-io/meta-rust` where as with `meta-rust-meta-rust` it is 1.0.2b. Maybe I can validate this if I know what is causing `libterm` issue by using `meta-rust/meta-rust` just with libterm issue fix

derekstraka, 2016-06-29:
@kteza1 - If I had to guess, I would say that there is a race condition on when libgit2-native is getting built causing things to not get built with SSL support.  To humor me, you could try a `bitbake -c cleansstate libgit2-native cargo-native;` and then do the rust build again to see if that fixes it?

kteza1, 2016-06-29:
@derekstraka 😸 😈 I think you are right. It worked. I was stuck with this error for quite some time

derekstraka, 2016-06-29:
Cool.  I can look at adding a bbappend for the DEPENDS as a temp fix, but I'll need to get that solved upstream.

kteza1, 2016-06-29:
@derekstraka Awesome. Thanks a lot. What might be the reason for `libterm` symbol lookup error?

cardoe, 2016-07-30:
The libterm issue is still an unknown to me.

cardoe, 2016-08-03:
@kteza1 Do you want to try #63 and confirm it resolves the issue for you?

kteza1, 2016-08-03:
I'll confirm by tomorrow @cardoe

kteza1, 2016-08-04:
@cardoe Seems to be fixed :+1: . I think you've done a lot of clean up. May I know what the problem was?

cardoe, 2016-08-04:
@kteza1 It was actually @derekstraka that did the work. I believe it was 7c494d94f0100d56d205e0bc6eb39454f29ebb3b that fixed the issue however.

PR #63 (2016-08-04): Miscellaneous cleanups for support recipes
* Remove recipes and files that an no longer required
* Update libgit2 to prepare for submission to meta-openembedded
* Fix a bug in cargo_util where the directory may not yet exist prior to touching .cargo-lock-index

Resolves #53
closed 2016-08-04 by cardoe