meta-rust/meta-rust #54: Yocto versioned branches

cardoe, 2016-06-29:
So meta-rust always had the development happening on a master branch but for work I need to be able to ship things based on Yocto release branches (e.g. fido, jethro, kergoth) so I'd like to push those branches into this repo now. Previously they've been here:

https://github.com/starlab-io/meta-rust/tree/fido
https://github.com/starlab-io/meta-rust/tree/jethro

They diverged from meta-rust a little bit but now that we're all working together on this project I see future named branches just forking from master. Sound ok to everyone? @jmesmon @derekstraka @srwalter

srwalter, 2016-06-29:
That sounds reasonable to me

derekstraka, 2016-06-29:
Works for me

cardoe, 2016-06-29:
Pushed.

closed 2016-06-29 by cardoe