meta-rust/meta-rust #55: Update README to reflect support for multiple yocto distros

jmesmon, 2016-06-29:
@cardoe @derekstraka @srwalter 

Right now we still have the old text from when I was being lazy and only dealing with master. I'm not sure precisely what should be put in it's place, perhaps it could just be removed?

[I'll also trim some of the development branches that were left over from my tree and shouldn't matter to most folks]

srwalter, 2016-07-05:
How about this wording?

"Generally, the master branch supports the latest master of poky.  When poky creates releases, we will created a branch with the same name as the poky release, which should always work with that poky release.  However, these branches will generally be less tested than the master branch."

jmesmon, 2016-07-11:
I've switched to a slight variation on the wording provided by @srwalter.

In the future we may also want to specify responsibility of release branches as well.

commit 1b78a658 (2016-07-11): Update README.md

Fixes #55
commit ff7c0827 (2016-07-12): Update README.md

Fixes #55
closed 2016-07-11 by jmesmon