meta-rust/meta-rust #59: Add krogoth branch

hmvp, 2016-07-26:
Yocto 2.1 is the current stable release. It would be nice if meta-rust had a krogoth branch to support 2.1

cardoe, 2016-07-30:
We're currently working on krogoth in master right now. Once krogoth support is working well, we'll make a branch.

cc @derekstraka

derekstraka, 2016-08-18:
@cardoe - Should we create a krogoth branch or did you want to wait for an updated rust version?

derekstraka, 2016-08-18:
@cardoe - I validated the master build.  The only issue I ran into was the random number blocking.  We will probably want to pull that patch over while we use rust 1.7

cardoe, 2016-08-18:
I think we just make a krogoth branch now but I agree let's backport the rng fix

cardoe, 2016-08-19:
krogoth branch has been pushed

closed 2016-08-19 by cardoe