meta-rust/meta-rust #62: Update libgit2 and push into meta-openembedded

derekstraka, 2016-08-01:
The libgit2 recipes can be updated to the latest version and pushed into meta-openembedded.  They can then be removed from meta-rust.  There are a few other consumers of the library that would benefit from it being in meta-openembedded instead of maintained in this layer.

derekstraka, 2016-08-01:
It looks like the meta-rust common license can also be removed when the libgit2 recipes are moved

derekstraka, 2016-08-02:
Patch submitted to the [meta-openembedded ML](http://lists.openembedded.org/pipermail/openembedded-devel/2016-August/108422.html)

cardoe, 2016-08-06:
Looks like v2 is [posted here](http://lists.openembedded.org/pipermail/openembedded-devel/2016-August/108445.html).

derekstraka, 2016-08-15:
This has been merged into meta-openembedded master, so I'll create a PR once krogoth gets created in #59

cardoe, 2016-08-19:
krogoth has been created

cardoe, 2016-09-06:
Closed via #75

closed 2016-09-06 by cardoe