meta-rust/meta-rust #65: MACHINE="qemuarm" + TCLIB="musl" results in rust-hello-world failing to run

jmesmon, 2016-08-08:
```
root@qemuarm:~# RUST_BACKTRACE=1 rust-hello-world 
thread '<unnamed>' panicked at 'failed to allocate a guard page', /home/x/obj/y/tmp/work-shared/rust-1.7.0-r0/rustc-1.7.0/src/libstd/sys/unix/thread.rs:252
stack backtrace:
   1:    0x19c23 - <unknown>
   2:    0x19c23 - <unknown>
   3:    0x19c23 - <unknown>
[... repeats the same thing to 100 frames ...]
```

- poky: `dfc016fbf13e62f7767edaf7abadf1d1b72680b2`
- meta-openembedded: `7e4f22517ada29f634970721481f0a7de1f3bffd`
- meta-rust: latest + #64 

local.conf has this appended to it (I'm testing with gcc 6 as I type this):
```
GCCVERSION = "5.%"
MACHINE = "qemuarm"
TCLIB = "musl"
IMAGE_INSTALL_append = " rust-hello-world"

BUILD_CXXFLAGS_append = " -D_GLIBCXX_USE_CXX11_ABI=0"
TARGET_CXXFLAGS_append = " -D_GLIBCXX_USE_CXX11_ABI=0"
CXXFLAGS_append = " -D_GLIBCXX_USE_CXX11_ABI=0"

BUILD_CXXFLAGS_remove_pn-gcc-runtime = "-D_GLIBCXX_USE_CXX11_ABI=0"
TARGET_CXXFLAGS_remove_pn-gcc-runtime = "-D_GLIBCXX_USE_CXX11_ABI=0"
CXXFLAGS_remove_pn-gcc-runtime = "-D_GLIBCXX_USE_CXX11_ABI=0"
```

Image is core-image-minimal.

running via `runqemu qemuarm serial`.

jmesmon, 2016-08-09:
`GCCVERSION = "6.%"` results in the same behavior.

derekstraka, 2016-08-18:
Does this work when it is a non-musl target?  I just want to make sure it isn't something funny with the arm9ejs target they use for qemuarm

derekstraka, 2016-08-18:
This looks rust issue looks interesting: Failed to allocate a guard page [#32082](https://github.com/rust-lang/rust/issues/32082)

jmesmon, 2016-08-19:
@derekstraka I'm doing a rebuild with `TCLIBC` commented (so it uses the non-musl default). I'll let you know what I find.

WRT rust issue: looks like part of it was a static vs dynamic linking problem, and the other was an unspecfied issue with the target spec (? I'm not quite sure). Without digging further, we might be hitting either of those problems.

derekstraka, 2016-08-19:
@jmesmon: It also looks like rust-1.7.0 doesn't build with some of the gcc6 additions.  I'll look at adding a patch for that unless you already have something

jmesmon, 2016-08-19:
I didn't have to make any modifications to convince gcc6 and rust to play nice. I suppose I might have accidentally preserved some bits from the gcc5 build... :(

jmesmon, 2016-08-19:
Just tried out non-musl: works fine:

```
root@qemuarm:~# rust-hello-world 
Hello, world!
root@qemuarm:~#
```

jmesmon, 2016-08-19:
Perhaps https://github.com/rust-lang/rfcs/pull/1721 could resolve our static vs dynamic issue in musl (presuming that is the problem).

jmesmon, 2016-09-25:
I've retested with my changes in #85, and this seems to work ok. It seems likely it was fixed by something else (like the rustc update). In any case, this can definitely be closed after #85 is merged.

PR #93 (2016-10-02): Ensure proper CXXFLAGS & CFLAGS are passed to rust's `rustllvm` component (c++ files built as part of rustc)
Fixes #47 

Fixes #65  (though might have actually been fixed earlier).

Also includes a few other cleanups.

These changes are extracted from #85 , just removing the commit that merges the building of rust's llvm back into rust itself.
closed 2016-10-02 by cardoe