meta-rust/meta-rust #69: Compilation of rust-cross fails for QEMUx86-64

leon-anavi, 2016-09-01:
Hi,

I noticed that do_compile of recipe rust-cross.bb for QEMUx86-64 fails with the following errors due to undefined reference from LLVM:
```
| note: x86_64-linux/stage0/lib/x86_64-poky-linux/rustlib/x86_64-linux/lib/librustc_llvm-e8edd0fd.so: undefined reference to `llvm::TargetRegistry::lookupTarget(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
| x86_64-linux/stage0/lib/x86_64-poky-linux/rustlib/x86_64-linux/lib/librustc_llvm-e8edd0fd.so: undefined reference to `llvm::RTDyldMemoryManager::getSymbolAddressInProcess(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
| x86_64-linux/stage0/lib/x86_64-poky-linux/rustlib/x86_64-linux/lib/librustc_llvm-e8edd0fd.so: undefined reference to `llvm::sys::DynamicLibrary::getPermanentLibrary(char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
| x86_64-linux/stage0/lib/x86_64-poky-linux/rustlib/x86_64-linux/lib/librustc_llvm-e8edd0fd.so: undefined reference to `llvm::createPrintModulePass(llvm::raw_ostream&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
| x86_64-linux/stage0/lib/x86_64-poky-linux/rustlib/x86_64-linux/lib/librustc_llvm-e8edd0fd.so: undefined reference to `llvm::report_fatal_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
| x86_64-linux/stage0/lib/x86_64-poky-linux/rustlib/x86_64-linux/lib/librustc_llvm-e8edd0fd.so: undefined reference to `llvm::SectionMemoryManager::finalizeMemory(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
| x86_64-linux/stage0/lib/x86_64-poky-linux/rustlib/x86_64-linux/lib/librustc_llvm-e8edd0fd.so: undefined reference to `llvm::Triple::normalize[abi:cxx11](llvm::StringRef)'
| x86_64-linux/stage0/lib/x86_64-poky-linux/rustlib/x86_64-linux/lib/librustc_llvm-e8edd0fd.so: undefined reference to `llvm::RTDyldMemoryManager::getPointerToNamedFunction(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
| collect2: error: ld returned 1 exit status
```

I am using the latest version of the master branch on meta-rust with Poky release Krogoth 2.1.

Best regards, Leon

jmesmon, 2016-09-01:
Looks like a dup of #47.

closed 2016-09-01 by jmesmon

leon-anavi, 2016-09-02:
Hi @jmesmon,

Thank you for the provided suggestion. I have applied the described workaround to conf/local.conf and I successfully build rustc, cargo and rust-hello-world in Poky image on PC with Ubuntu 16.04 and gcc version 5.4.0.

Best regards, Leon