meta-rust/meta-rust #73: Need general guidelines for committers

jmesmon, 2016-09-06:
Different folks have different expectations, we should probably write down some simple guidelines to follow when making changes so that everyone is mostly-happy.

Probably something that could go in the README or some other document.

@cardoe @derekstraka @srwalter

jmesmon, 2016-09-06:
From #71 it sounds like there is a preference to have anyone-else do the actual merge (at least for larger changes). I'm OK with this being the guideline for now, but it'd be nice to write it down and do a bit of refinement.

Just to give where I'm coming from with this:

 - at the moment, I'm not using meta-rust for any projects, so I'm not doing builds with much frequency.
 - I'm not entirely comfortable with personally merging things that I haven't build tested (I'd be comfortable merging if we had some automated testing), so at the moment I typically leave merging riskier looking things to others.

I think it would be useful to note in these guidelines:

 - what level of build-testing is expected of submitters and/or mergers (we should consider that having lots of this is not good due to being a bit of work/time due to our lack of automation).
 - what configurations shouldn't have their build broken
 - what level of run-testing (if any)
 - who does the merge

cardoe, 2016-09-06:
My comments were more along the line of another person doing the merge signifies that there was at least one other person happy with the changes that it was merged. Its also certainly appropriate for someone to merge their own changes if someone else says "👍 feel free to merge". Just looking for a clear indication of peer review.

cardoe, 2016-12-29:
I added some notes at https://github.com/meta-rust/meta-rust/wiki if that sounds ok then we can close this issue.

jmesmon, 2016-12-31:
That all looks ok to me

closed 2016-12-31 by jmesmon