meta-rust/meta-rust #74: Project automation

jmesmon, 2016-09-06:
- [ ] auto-assign PRs to someone with commit access
- [ ] auto-assign Issues so someone is pinged to glance at them
- [ ] build testing (travis? buildbot host? something else?)
- [ ] bors/homu/etc like setup for merging PRs.

derekstraka, 2016-09-06:
IIRC the travis builds we were originally working on timed out because it took so long to build all of the rust items.

cardoe, 2016-11-09:
So I setup some more protected branches (for each of the Yocto releases) and for master set the requirement that the PR passed through Jenkins and is up to date with master.

cardoe, 2016-12-04:
Played around with homu a bit. Its dependent on Travis, which won't work for us due to the build times. bors is the closed source version of homu.

jmesmon, 2016-12-04:
bors is (IIRC), the old version: https://github.com/graydon/bors

There is also a re-write of homu in rust somewhere (I'm looking), but I'm not sure who is using it.