meta-rust/meta-rust #77: The recipe rust-cross-x86_64 is trying to install files into a shared area when those files already exist.

alexhumphreys, 2016-09-09:
I'm upgrading from 1.7 to 1.10 and I get the following error:

```
DEBUG: Python function do_populate_sysroot finished
DEBUG: Executing python function do_qa_staging
NOTE: QA checking staging
DEBUG: Python function do_qa_staging finished
DEBUG: Executing python function sstate_task_postfunc
DEBUG: Staging files from /opt/pro/poky/build/tmp/work/x86_64-linux/rust-cross-x86_64/1.10.0-r0/sysroot-destdir/opt/pro/poky/build/tmp/sysroots/x86_64-linux to /opt/pro/poky/build/tmp/sysroots/x86_64-linux
ERROR: The recipe rust-cross-x86_64 is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json
 Matched in manifest-x86_64-rust-native.populate_sysroot
Please verify which recipe should provide the above files.
The build has stopped as continuing in this scenario WILL break things, if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error however switching DISTRO_FEATURES on an existing build directory is not supported, you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
DEBUG: Python function sstate_task_postfunc finished
ERROR: Function failed: sstate_task_postfunc
```

I destroyed my `tmp/` dir, got same error, then removed the entire `build/` dir, still the same error.

If i run `find tmp/ -name x86_64-linux.json` I get the following:

```
tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json
tmp/work/x86_64-linux/rust-cross-x86_64/1.10.0-r0/sysroot-destdir/opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json
tmp/work/x86_64-linux/rust-cross-x86_64/1.10.0-r0/image/opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json
tmp/work/x86_64-linux/rust-cross-x86_64/1.10.0-r0/targets/x86_64-linux.json
tmp/work/x86_64-linux/rust-native/1.10.0-r0/sysroot-destdir/opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json
tmp/work/x86_64-linux/rust-native/1.10.0-r0/image/opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json
tmp/work/x86_64-linux/rust-native/1.10.0-r0/targets/x86_64-linux.json
```

`tmp/work/x86_64-linux/rust-native/1.10.0-r0/sysroot-destdir/opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json` Seems to be the one that is there at `tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json` first. 

Then it seems bitbake tries to write `tmp/work/x86_64-linux/rust-cross-x86_64/1.10.0-r0/sysroot-destdir/opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib/x86_64-linux.json` to the same location, causing the error.

Here's the [rust-cross json file](https://gist.github.com/alexhumphreys/8b4769b9811ed258cddb5165c949cb36), and here's the [rust-native json file](https://gist.github.com/alexhumphreys/c466d4fec74b4201d8b43c853f589a2e). The only difference seems to be rust-cross json file has an extra `"-Wl,-rpath=../../lib"` in the `post-link-args` array.

So yeah, if you have any idea how to resolve this it'd be greatly appreciated.

Cheers!

jmesmon, 2016-09-09:
@srwalter perhaps we should generate a cross rustlib dir with the host libraries symlinked in to avoid this? Or have -cross only generate the target's target spec?

@alexhumphreys are there any other target specs in rustlib? This may be caused by TARGET_SYS and BUILD_SYS/HOST_SYS being the same value (knowing the value of those env vars would also help us).

alexhumphreys, 2016-09-13:
Here's the build configuration:
```
Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Debian-8.5"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "genericx86-64"
DISTRO            = "poky-ats"
DISTRO_VERSION    = "0.314"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
```

I'm afraid I don't know what you mean by "target specs in rustlib", could you elaborate for me?

jmesmon, 2016-09-14:
@alexhumphreys 
`target specs in rustlib`:
```
ls tmp/work/x86_64-linux/rust-cross-x86_64/1.10.0-r0/sysroot-destdir/opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib
```

But it looks like our issue is that native & cross both generate target specs for native (`x86_64_linux.json`) and they differ slightly.

Simple solution is probably to have rust-cross stop generating a native target spec.

alexhumphreys, 2016-09-15:
Not sure if you still need it, but for the sake of completeness:

```
ls tmp/work/x86_64-linux/rust-cross-x86_64/1.10.0-r0/sysroot-destdir/opt/pro/poky/build/tmp/sysroots/x86_64-linux/usr/lib/rustlib
x86_64-linux.json  x86_64-poky-linux.json
```

I tried to build with #79 and it successfully built the rust recipes. So at least to me it looks good.

PR #79 (2016-09-15): rust-cross: fix build where TARGET_ARCH and HOST_ARCH are the same
By copying exactly the target we intend to.

Fixes #77 

Testing with new enough poky may also need #78
commit 237a22fd (2016-09-15): rust-cross: fix build where TARGET_ARCH and HOST_ARCH are the same

By copying exactly the target we intend to.

Fixes #77
closed 2016-09-15 by cardoe