meta-rust/meta-rust #80: re-merge rust-llvm-native & rust-native

jmesmon, 2016-09-14:
Now that we build std for target instead of rebuilding the whole compiler, splitting out rust-llvm doesn't save us any build time.

Should we merge them back together?

cardoe, 2016-09-15:
I agree

cardoe, 2016-11-02:
So I'll rescind my "I agree" only because its nice for me to save some time in rebuilding when I get `rust-llvm` working but still don't have `rust` working.

jmesmon, 2016-11-02:
Yep, after actually trying this out in #85, it seems like it's actually more work (due to needing to have our configuration for llvm's build process work it's way through rust's build process) to merge these two together. Also, keeping these separate could make it more straight forward to depend on an actual upstream llvm in the future.

I'll close this for now.

closed 2016-11-02 by jmesmon