meta-rust/meta-rust #81: Is it possible to build rustc and cargo for target?

kteza1, 2016-09-19:
I tried using rustup to be able to directly compile on target board and it didn't work. Is it possible to build rustc and cargo using meta-rust?

cardoe, 2016-09-19:
In theory yes, in practice I'm not sure if anyone has actually tested that.

kteza1, 2016-09-19:
I'll test that. All I've to do is include `rustc` and `cargo` to image bb file right?

cardoe, 2016-09-19:
Should be

cardoe, 2016-10-31:
@kteza1 I'm assuming this has not worked for you? I just did `bitbake rust-llvm` and it failed to build for me.

kteza1, 2016-11-01:
@cardoe Oops sorry for not getting back. I tried this some time back and it failed I think. I'll report back if I try this again.

jmesmon, 2016-11-01:
I've tried to do this too with the experiment in merging the rust-llvm build into rust. IIRC, it failed there too due to llvm not recognizing the target it was being built for. It likely needs to be passed an llvm-flavor target instead of a OE target.

cardoe, 2016-11-04:
So in my branch to build rust-llvm with cmake I've managed to get it compiled for the target. I haven't tried to build rust with it yet. But figured I'd provide an update.

cardoe, 2016-11-04:
While #105 fixes rust-llvm it does not fix rust which still fails with the following:

```
-isystem/var/build/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -D_GLIBCXX_USE_CXX11_ABI=0  --sysroot=/var/build/build/tmp/sysroots/qemux86-64  -I /var/build/build/tmp/sysroots/x86_64-linux/usr/include -I /var/build/build/tmp/work/core2-64-poky-linux/rust/1.10.0-r0/rustc-1.10.0/src/rustllvm/include /var/build/build/tmp/work/core2-64-poky-linux/rust/1.10.0-r0/rustc-1.10.0/src/rustllvm/RustWrapper.cpp
In file included from /var/build/build/tmp/sysroots/x86_64-linux/usr/include/llvm/ADT/Hashing.h:48:0,
                 from /var/build/build/tmp/sysroots/x86_64-linux/usr/include/llvm/ADT/ArrayRef.h:13,
                 from /var/build/build/tmp/sysroots/x86_64-linux/usr/include/llvm/IR/IRBuilder.h:18,
                 from /var/build/build/tmp/work/core2-64-poky-linux/rust/1.10.0-r0/rustc-1.10.0/src/rustllvm/rustllvm.h:11,
                 from /var/build/build/tmp/work/core2-64-poky-linux/rust/1.10.0-r0/rustc-1.10.0/src/rustllvm/ExecutionEngineWrapper.cpp:11:
/var/build/build/tmp/sysroots/x86_64-linux/usr/include/llvm/Support/DataTypes.h:35:17: fatal error: cmath: No such file or directory
 #include <cmath>
```

srwalter, 2016-11-07:
If this is a build failure, then you'd need to DEPEND on gcc-runtime, rather RDEPEND, right?

cardoe, 2016-12-02:
I do not believe anyone is working on this but we would certainly accept patches for this support.

macpijan, 2019-05-16:
Currently (I am on 0621067c560f6e3e1a4c4c08ac173393bbafad5c commit), the error message is:

```
Sstate summary: Wanted 5 Found 0 Missed 5 Current 92 (0% match, 94% complete)                                                                                                                                        
NOTE: Executing SetScene Tasks                                                                                                                                                                                       
NOTE: Executing RunQueue Tasks                                                                                                                                                                                       
ERROR: rust-1.33.0-r0 do_compile: Function failed: do_compile (log file is located at /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/temp/log.do_compile.224)
ERROR: Logfile of failure stored in: /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/temp/log.do_compile.224
Log data follows:                                    
| DEBUG: Executing shell function do_compile                                                                                                                                                                         
| COMPILE rust build                      
| running: /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/rust-snapshot/bin/cargo build --manifest-path /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/rustc-1.33.0-src/src/bootstrap/Cargo.toml --f$
ozen
|     Finished dev [unoptimized] target(s) in 7.95s
| running: /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/rustc-1.33.0-src/build/bootstrap/debug/bootstrap build --verbose
|                                    
|                                 
| failed to execute command: "/work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/recipe-sysroot/usr/lib/llvm-rust/bin/llvm-config" "--bindir"
| error: No such file or directory (os error 2)
|                                
|                             
| Traceback (most recent call last):
|   File "src/bootstrap/bootstrap.py", line 860, in <module>
|     main()                                                                    
|   File "src/bootstrap/bootstrap.py", line 843, in main                
|     bootstrap(help_triggered)                                          
|   File "src/bootstrap/bootstrap.py", line 829, in bootstrap          
|     run(args, env=env, verbose=build.verbose)
|   File "src/bootstrap/bootstrap.py", line 141, in run
|     raise RuntimeError(err)                                         
| RuntimeError: failed to run: /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/rustc-1.33.0-src/build/bootstrap/debug/bootstrap build --verbose
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/temp/log.do_compile.224)
ERROR: Task (/work/meta-rust/recipes-devtools/rust/rust_1.33.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 596 tasks of which 595 didn't need to be rerun and 1 failed.
```

I am going to spend a while figuring this out

david-mcgillicuddy-moixa, 2019-08-30:
Hi @macpijan did you ever figure this out? I'm getting exactly this error when trying to run the build on an AWS codebuild server.

macpijan, 2019-08-30:
@david-mcgillicuddy-moixa Sadly no, I had to finish that project and shipped prebuild rustc compiler on the target.

david-mcgillicuddy-moixa, 2019-08-30:
My case might just have been an OOM actually, trying again with more seemed to work.

macpijan, 2019-08-30:
Interesting. Are you building the rustc compiler to be installed on the target board, or something else? What is the target arch?

david-mcgillicuddy-moixa, 2019-08-30:
I didn't set the build up, but it's:
```
BUILD_SYS = "x86_64-linux" 
TARGET_SYS = "arm-oe-linux-gnueabi" 
```
Precompiled rustc binaries would probably work for our use case too - they are not installed on the target board, only the artifact image.

wdehoog, 2019-10-21:
I tried to build cargo for arm target and got:
```
error[E0463]: can't find crate for `proc_macro`
  --> /usr/src/debug/cargo/1.37.0-r0/rustc-1.37.0-src/vendor/proc-macro2/src/lib.rs:87:1
   |
87 | extern crate proc_macro;
   | ^^^^^^^^^^^^^^^^^^^^^^^^ can't find crate

```

SrijanNandi, 2020-12-16:
Same issue building rust-1.46.0-r0 on x86_64 for a x86_64 target.

|    Compiling bootstrap v0.0.0 (/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.46.0-r0/rustc-1.46.0-src/src/bootstrap)
|     Finished dev [unoptimized + debuginfo] target(s) in 1m 05s
| running: /opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.46.0-r0/rustc-1.46.0-src/build/bootstrap/debug/bootstrap -j 10 build --verbose
| 
| 
| failed to execute command: "/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.46.0-r0/recipe-sysroot/usr/lib/llvm-rust/bin/llvm-config" "--bindir"
| error: No such file or directory (os error 2)
| 
| 
| Traceback (most recent call last):
|   File "src/bootstrap/bootstrap.py", line 998, in <module>
|     main()
|   File "src/bootstrap/bootstrap.py", line 981, in main
|     bootstrap(help_triggered)
|   File "src/bootstrap/bootstrap.py", line 967, in bootstrap
|     run(args, env=env, verbose=build.verbose)
|   File "src/bootstrap/bootstrap.py", line 143, in run
|     raise RuntimeError(err)
| RuntimeError: failed to run: /opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.46.0-r0/rustc-1.46.0-src/build/bootstrap/debug/bootstrap -j 10 build --verbose
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/opt/grays-poky-dunfell/meta-rust/recipes-devtools/rust/rust_1.46.0.bb:do_compile) failed with exit code '1'

SrijanNandi, 2021-02-21:
Same issue building rust-1.49.0-r0 on x86_64 for a x86_64 target.

Log data follows:
| DEBUG: Executing shell function do_compile
| COMPILE rust build --stage 2
| running: /opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/rust-snapshot/bin/cargo build --manifest-path /opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/rustc-1.49.0-src/src/bootstrap/Cargo.toml --frozen
|     Finished dev [unoptimized + debuginfo] target(s) in 0.34s
| running: /opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/rustc-1.49.0-src/build/bootstrap/debug/bootstrap -j 10 build --stage 2 --verbose
| finding compilers
| CC_x86_64-unknown-linux-gnu = "/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/wrapper/build-rust-cc"
| CFLAGS_x86_64-unknown-linux-gnu = ["-ffunction-sections", "-fdata-sections", "-fPIC", "-m64", "-isystem/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/recipe-sysroot-native/usr/include", "-pipe"]
| CXX_x86_64-unknown-linux-gnu = "/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/wrapper/build-rust-cxx"
| CXXFLAGS_x86_64-unknown-linux-gnu = ["-ffunction-sections", "-fdata-sections", "-fPIC", "-m64", "-isystem/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/recipe-sysroot-native/usr/include", "-pipe"]
| AR_x86_64-unknown-linux-gnu = "x86_64-poky-linux-gcc-ar"
| CC_x86_64-poky-linux = "/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/wrapper/target-rust-cc"
| CFLAGS_x86_64-poky-linux = ["-ffunction-sections", "-fdata-sections", "-fPIC", "-m64", "-pipe", "-g", "-feliminate-unused-debug-types", "-fmacro-prefix-map=/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0=/usr/src/debug/rust/1.49.0-r0", "-fdebug-prefix-map=/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0=/usr/src/debug/rust/1.49.0-r0", "-fdebug-prefix-map=/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/recipe-sysroot=", "-fdebug-prefix-map=/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/recipe-sysroot-native="]
| CXX_x86_64-poky-linux = "/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/wrapper/target-rust-cxx"
| CXXFLAGS_x86_64-poky-linux = ["-ffunction-sections", "-fdata-sections", "-fPIC", "-m64", "-pipe", "-g", "-feliminate-unused-debug-types", "-fmacro-prefix-map=/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0=/usr/src/debug/rust/1.49.0-r0", "-fdebug-prefix-map=/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0=/usr/src/debug/rust/1.49.0-r0", "-fdebug-prefix-map=/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/recipe-sysroot=", "-fdebug-prefix-map=/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/recipe-sysroot-native="]
| AR_x86_64-poky-linux = "x86_64-poky-linux-gcc-ar"
| running sanity check
| 
| 
| failed to execute command: "/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/recipe-sysroot/usr/lib/llvm-rust/bin/llvm-config" "--bindir"
| error: No such file or directory (os error 2)
| 
| 
| Traceback (most recent call last):
|   File "src/bootstrap/bootstrap.py", line 1099, in <module>
|     main()
|   File "src/bootstrap/bootstrap.py", line 1082, in main
|     bootstrap(help_triggered)
|   File "src/bootstrap/bootstrap.py", line 1068, in bootstrap
|     run(args, env=env, verbose=build.verbose)
|   File "src/bootstrap/bootstrap.py", line 153, in run
|     raise RuntimeError(err)
| RuntimeError: failed to run: /opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/rustc-1.49.0-src/build/bootstrap/debug/bootstrap -j 10 build --stage 2 --verbose
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/opt/grays-poky-dunfell/build-grays-dunfell/tmp/work/core2-64-poky-linux/rust/1.49.0-r0/temp/run.do_compile.2593' failed with exit code 1:

kraj, 2021-02-21:
this is known limitation

SrijanNandi, 2021-02-22:
Any workaround to get rust working on x86_64

ramumr06, 2022-10-10:
> Any workaround to get rust working on x86_64

Hi SrijanNandi,
I am getting  llvm-rust/bin/llvm-config" "--bindir" | error: No such file or directory (os error 2).
Host System : x86-64
Target : ARM
Do you know any work around for this ??

ramumr06, 2022-10-10:
> Currently (I am on [0621067](https://github.com/meta-rust/meta-rust/commit/0621067c560f6e3e1a4c4c08ac173393bbafad5c) commit), the error message is:
> 
> ```
> Sstate summary: Wanted 5 Found 0 Missed 5 Current 92 (0% match, 94% complete)                                                                                                                                        
> NOTE: Executing SetScene Tasks                                                                                                                                                                                       
> NOTE: Executing RunQueue Tasks                                                                                                                                                                                       
> ERROR: rust-1.33.0-r0 do_compile: Function failed: do_compile (log file is located at /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/temp/log.do_compile.224)
> ERROR: Logfile of failure stored in: /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/temp/log.do_compile.224
> Log data follows:                                    
> | DEBUG: Executing shell function do_compile                                                                                                                                                                         
> | COMPILE rust build                      
> | running: /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/rust-snapshot/bin/cargo build --manifest-path /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/rustc-1.33.0-src/src/bootstrap/Cargo.toml --f$
> ozen
> |     Finished dev [unoptimized] target(s) in 7.95s
> | running: /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/rustc-1.33.0-src/build/bootstrap/debug/bootstrap build --verbose
> |                                    
> |                                 
> | failed to execute command: "/work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/recipe-sysroot/usr/lib/llvm-rust/bin/llvm-config" "--bindir"
> | error: No such file or directory (os error 2)
> |                                
> |                             
> | Traceback (most recent call last):
> |   File "src/bootstrap/bootstrap.py", line 860, in <module>
> |     main()                                                                    
> |   File "src/bootstrap/bootstrap.py", line 843, in main                
> |     bootstrap(help_triggered)                                          
> |   File "src/bootstrap/bootstrap.py", line 829, in bootstrap          
> |     run(args, env=env, verbose=build.verbose)
> |   File "src/bootstrap/bootstrap.py", line 141, in run
> |     raise RuntimeError(err)                                         
> | RuntimeError: failed to run: /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/rustc-1.33.0-src/build/bootstrap/debug/bootstrap build --verbose
> | WARNING: exit code 1 from a shell command.
> | ERROR: Function failed: do_compile (log file is located at /work/build/tmp/work/aarch64-poky-linux/rust/1.33.0-r0/temp/log.do_compile.224)
> ERROR: Task (/work/meta-rust/recipes-devtools/rust/rust_1.33.0.bb:do_compile) failed with exit code '1'
> NOTE: Tasks Summary: Attempted 596 tasks of which 595 didn't need to be rerun and 1 failed.
> ```
> 
> I am going to spend a while figuring this out

Hi macpijan
I am getting llvm-rust/bin/llvm-config" "--bindir" | error: No such file or directory (os error 2).
Host System : x86-64
Target : ARM
Do you know any work around for this ??

rwmacleod, 2022-11-25:
It's possible using oe-core/langdale or oe-core/master. I haven't tried it yet, have you? If so and it works well for you then please consider closing this bug. I'm hoping to get to that this weekend.

rwmacleod, 2022-11-27:
cargo doesn't build for the target  https://bugzilla.yoctoproject.org/show_bug.cgi?id=14975

Rust does build but fails for me with:

```
root@qemux86-64:~# cat /tmp/hw.rs
fn main() {
    println!("Hello, world!")
}

root@qemux86-64:~# rustc /tmp/hw.rs
error: Error loading target specification: Could not find specification for target "x86_64-poky-linux-gnu". Run `rustc --print target-list` for a list of built-in targets

root@qemux86-64:~# rustc --print target-list
error: Error loading target specification: Could not find specification for target "x86_64-poky-linux-gnu". Run `rustc --print target-list` for a list of built-in targets
```

rwmacleod, 2022-11-27:
If you manually copy
   tmp/work/core2-64-poky-linux/rust/1.65.0-r0/rustc-1.65.0-src/src/test/run-make-fulldeps/rustdoc-target-spec-json-path/target.json
to:
   /usr/lib/rustlib/x86_64-poky-linux-gnu/target.json
and have this in your local.conf:
IMAGE_INSTALL:append = " rust packagegroup-core-buildessential"

it works. Patch coming for master and langdale soon.

rwmacleod, 2022-12-25:
Yes, fixed in oe-core/master:
https://git.openembedded.org/openembedded-core/commit/?id=b187185ceecf1c852b1fdc47451d157a41962d69
Please try that out and close this issue.