meta-rust/meta-rust #82: rust 1.10 for jethro branch

kteza1, 2016-09-19:
Hi. Any plans for bumping version on jethro branch?

jmesmon, 2016-09-19:
- I'm not aware of anyone working on that
- At the moment, we don't have a specific policy around upgrading release branches, but I feel like it'd make sense to accept patches that update them. There is a bit of a question around what level of compat should be kept in the release branches.

derekstraka, 2016-09-19:
For yocto, typically only bug fixes go into the stable branches rather than version updates.  Jethro will probably only be actively supported for another couple months until 2.2 comes out in October (https://wiki.yoctoproject.org/wiki/Releases), and then it would move to community support.

That said, with how fast rust moves, I wouldn't be opposed to moving forward as long as the PREFERRED_VERSION is set to the original and the previous version is retained to make sure it doesn't break people using the previous 'stable' version.

cardoe, 2016-09-25:
So I think end result is that we'll accept a patch to do this as long as retain the existing version and set PREFERRED_VERSION to the old version to not break people that have packages that won't build with 1.10. However none of the current maintainers are working on this so it must be a supplied PR. @kteza1 Are you planning on providing this PR?

jmesmon, 2016-09-25:
Would it not be more correct to use `DEFAULT_PREFERENCE = "-1"` for preventing automatic selection of newer versions?

cardoe, 2016-11-09:
So I'm honestly not sure how we'd backport this stuff to jethro without nuking 1.7 since the state of the versions results in something that can't be added alongside each other since the bbclasses conflict since we've changed them significantly and kept the same name (if you recall the `cargo_util` bbclass for example got renamed to `cargo` but there's more than that.)

cardoe, 2016-12-02:
any @meta-rust/maintainers want to take charge on this or is this a patches welcome request?

derekstraka, 2017-03-08:
Jethro has moved to community support only and I don't envision anyone doing this backport

closed 2017-03-08 by derekstraka