meta-rust/meta-rust #87: Remove internet access in do_compile for rust and cargo

tardyp, 2016-09-27:
Doing internet access in do_compile is not a good play with yocto

- prevent sstate, and local_mirrors to work correctly
- does not work with corporate proxies, as http_proxy variable is only available in do_fetch

I first created #83 as a workaround for the proxy issue, but know I think it was not a good idea for the above reason.

Creating this as an issue so that it is easier for downstream user to understand the situation and help.

cardoe, 2016-09-27:
So this is how this repo works. You just need to properly create recipes using https://crates.io/crates/cargo-bitbake to generate a proper `SRC_URI` variable to fetch down all the dependencies. Cargo does not have a way to tell it to never reach out to the internet to fetch dependencies currently. All we can do is fetch them down for Cargo and then let it run with all the files in place. Future versions of Cargo will improve the situation (patches have landed for the version to be released in 2 days) but will not make it bullet proof. Those changes won't come down the pipe for another 3 months.

I looked at the recipe that #84 mentions https://github.com/openivimobility/meta-oim/blob/master/recipes-oim/sota-client/sota-client.bb and its just completely wrong since its not including any dependencies. I would file a bug with that project.

tardyp, 2016-09-27:
@cardoe its not only about sota-client. 

Testing on the krogoth branch (15bb60780a7b7a97aec0e7e6b192a90c14b5fc7b), the cargo-native package itself will not build.

```
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 40 ARGS=--verbose
| /data/genivi-dev-platform/gdp-src-build/tmp/sysroots/x86_64-linux/usr/bin/rustc -V
| rustc 1.7.0
| target/snapshot/bin/cargo --version
| cargo 0.9.0-nightly (c4c6f39 2016-01-30)
| target/snapshot/bin/cargo build --target x86_64-linux --manifest-path /data/genivi-dev-platform/gdp-src-build/tmp/work/x86_64-linux/cargo-native/0.10.0-r0/cargo-0.10.0//Cargo.toml --release  --verbose
|     Updating registry `https://github.com/rust-lang/crates.io-index`
| failed to fetch `https://github.com/rust-lang/crates.io-index`
|
| Caused by:
|   [2/-1] Failed to connect to github.com: Connection timed out
| make: *** [cargo-x86_64-linux] Error 101
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /data/genivi-dev-platform/gdp-src-build/tmp/work/x86_64-linux/cargo-native/0.10.0-r0/temp/log.do_compile.19910)
```

I forgot to mention the downstring bug tracking the issue:
https://at.projects.genivi.org/jira/browse/GDP-404

tardyp, 2016-09-27:
So I retested on master branch, and I can see that indeed there is no more problem on cargo build (and I can see that the cargo.inc is filled with crate:// urls)

My build fails on sota-client which is expected.

I had to checkout the recipe-deps from the kergoth branch though, so master branch cannot work with kergoth oe as-is.

I quickly looked at porting crate-fetch to kergoth, but I fear it will essencially mean pulling everything from master branch.

@cardoe 
Could advice us on the best approach?

Shall we reopen, and make it clear the problem is in kergoth branch?

cardoe, 2016-09-27:
Ok I understand the issue now.

derekstraka, 2016-09-27:
That's fine.  I just don't want to get into the business of trying to pull them back to support jethro.

leon-anavi, 2016-09-27:
Hi @cardoe, @derekstraka, 

@tardyp reported that the corporate firewalls and this issue are blocking his local build of GENIVI Development Platform (GDP). 

Yocto/OE layer has been integrated in GDP and Automotive Grade Linux (AGL) because it is needed by RVI SOTA client and I am looking after it. 

I will highly appriciate if you could backport these Cargo improvements to branch Krogoth. After that I will update the references to meta-rust in both distributions and hopefully the issue will be solved. 

Best regards, Leon

cardoe, 2016-09-27:
Patches are always welcome.

cardoe, 2016-10-02:
I've pushed an update to `cargo-bitbake` to hopefully better generate the dependencies. I'd be curious to hear how you needed your recipe to look to get things to build with the `cargo_util` bbclass in the krogoth branch. Hopefully I can come up with a way to get `cargo-bitbake` to output that correctly.

cardoe, 2016-11-22:
So the only real path forward I see here is to actually make ANOTHER krogoth branch derived from master once I'm done with my master tweaks and add back in krogoth support. That's what I plan on doing for a work project where we use krogoth. Another option is to just blow away krogoth and replace it with that new krogoth branch. The new krogoth branch won't be done until I have Rust 1.12.1 in master since that's also what I need for work.

cardoe, 2016-12-01:
So @tardyp and @leon-anavi would it break you going to a newer Rust?

cardoe, 2016-12-01:
You can try testing with https://github.com/cardoe/meta-rust/tree/krogoth-1.12.1 where I'm putting together the changes.

cardoe, 2016-12-12:
This has been merged into the krogoth branch and should now work.

PR #129 (2016-12-12): Merge branch 'morty' into krogoth
Merges the changes in the morty branch into krogoth to get Rust 1.10.0
and Rust 1.12.1 into krogoth. Additionally this allows for Rust, Cargo
and any recipes to be built without internet access if they are defined
using cargo-bitbake. Fixes #87. Fixes #122.
closed 2016-12-12 by cardoe