meta-rust/meta-rust #88: document some type of practice for issue tagging (or not tagging)

jmesmon, 2016-09-27:
At the moment, I've occasionally created 'S-*' and 'R-*' tags for 'status' and 'releases' respectively. I'm not sure if there is some standard for these. Rust itself appears to use quite a lot of tags for their issues, but I'm not sure where (or if) the exact meaning of most of these is documented.

This is something we should probably add to the readme.

cardoe, 2016-12-02:
So I'd like to see the `R-yoctorelease` labels used on issues where they affect things. Then there's effectively 3 options for an issue:

- question
- enhancement
- bug

Further more `help wanted` asks other maintainers to weigh in. And the most recent `patch welcome` means none of the maintainers are working on this but a PR to add this feature would be accepted.

Do you want me to write up a markdown doc expressing that?

jmesmon, 2016-12-03:
Sure, something in the README or another doc with a quick overview could be useful.

To be fair, when I opened this, I was looking at rust's issue tagging, and trying to make sense of all the prefix letters. We probably don't have that level of complexity, though.

cardoe, 2016-12-08:
Would you rather a project wiki page instead of a markdown file in the project?

jmesmon, 2016-12-08:
That sounds reasonable.

cardoe, 2016-12-29:
https://github.com/meta-rust/meta-rust/wiki has been filled in.

closed 2016-12-29 by cardoe