meta-rust/meta-rust #94: Continuous Integration Testing

cardoe, 2016-10-02:
We need to get some form of CI for this project. In the past I tried to use Travis CI but bumped up against the 1 hour time limit every time.

derekstraka, 2016-10-25:
I've setup a jenkins server we can use - ci.asterius.io with a couple of slaves that can be used for CI.

cardoe, 2016-10-27:
@jmesmon Can you add the `metarustbot` user to the project and give it `repo` access?

jmesmon, 2016-10-27:
repo or organization?

I've sent invites for both. Note that the way things are configured right now, all organization members can create repositories, add new collaborators to existing repositories, and have read & write access to organization repositories (or at least, I believe that's what should happen based on what github's settings are telling me). It might make sense to use teams to limit this somewhat.

cardoe, 2016-10-27:
Well the `repo` access I needed was actually the API token permission so that was my confusion. But org or repo works. If we add other repos to this org (not sure if we will) then we can give it more access.

PR #96 (2016-10-31): Jenkins CI
This adds some scripts to build `rustfmt` using all of our tooling inside of a Docker container to ensure that all the dependencies are there. In theory anyone should be able to use this on their machine and confirm the same results. It should be possible to extend these scripts for building more items and potentially artifacting results for more tests. fixes #94
closed 2016-10-31 by derekstraka