meta-rust/meta-rust #95: rust shared source / package split

cardoe, 2016-10-26:
So I'll be real honest and admit that I don't understand why we do the shared source and split up so much of the configuration. I find the setup here a bit complex and hard to manage. In Gentoo where I maintain the Rust packages as well I've got everything split into a `rustllvm` and `rust`. I extract the tarball twice. I build the target mk files within the sources rather than try to provide an alternative directory. Effectively all I'm patching in Gentoo is the fetching of the stage0 tarball since I have the package manager do that and place it in the right location. What I'm getting at is can we refactor into a similar setup since to me conceptually it will be easier to make version bumps or what am I missing behind some of the separation and other patches.

/cc @jmesmon @derekstraka @srwalter

jmesmon, 2016-10-26:
IIRC, I originally did the source-split to avoid needing to have 3 separate copies of the source code around (`-native`, `-llvm`, `-cross`). As gcc was doing the same thing, I figured it might be a good idea to do so to reduce size/inodes/path-lookups/uncached-reads/extra-patch-applies. At the time I was also unsure if bitbake would properly avoid downloading the same source tarball 3 times over if something like this was not done.

Note that rebasing the patches is (from my perspective) a fairly easy part of updating meta-rust. In fact, that's most of what the WIP for 1.12 ( #89 ) has in it already (rebased patches). Hold up right now is spending time figuring out how to build new things for target (compiler-rt), and likely figuring out the cargo update. My feeling is the main complexity in updating meta-rust is handling cross compilation while strictly following the specification of the target (flags, features, etc) given by the OE metadata (rust tends to love it's magic configuration).

Also, IIRC, only 1 patch ( https://github.com/meta-rust/meta-rust/blob/master/recipes-devtools/rust/files/rust/0010-mk-allow-changing-the-platform-configuration-source-.patch ) is actually added for handling out-of-tree builds correctly.

One key patch that was (the code around it doesn't get updated much anymore, so it's a bit easier these days) a pain to maintain is the one fixing `libdir` and `bindir` (which I didn't upstream at the time because it would break windows, and it needs support for continuing to use rust's "magic libdir" strategy for default builds). This was absolutely required when we were building a `-cross` compiler, but is somewhat less crucial now as long as everyone uses default paths for everything. I'm not quite sure I want to start making that assumption.

cardoe, 2016-10-31:
Would you be upset if I dropped the shared source bits? There's just a lot of code machinery to make this happen and to me personally it adds to some maintenance burden even though we don't have to touch it all the time.

Does anyone else have an opinion? @posborne @srwalter @derekstraka ?

jmesmon, 2016-10-31:
I'm not too attached to the shared source pieces of the recipes.

I'd be interested in seeing how removing them affects (or doesn't affect) the build time & size. I'd imagine that it'd be a very small (if anything) change in overall time/size, but in that case I'm not sure why the gcc package would bother with the work of source-splitting.

cardoe, 2016-10-31:
I'll can do a few builds locally and compare.

posborne, 2016-10-31:
Dropping shared source is fine by me.

srwalter, 2016-10-31:
I could go either way.  I'm not married to shared-source, but if it's giving us a decent speed benefit I'd weigh on the side of keeping it.

cardoe, 2016-11-04:
Closing since #101 has merged.

closed 2016-11-04 by cardoe