meta-rust/meta-rust #102: remove recipes-core / recipes-graphics

cardoe, 2016-11-02:
The recipes in these directories are what I'll call "old style" BitBake recipes for dependencies. How do people feel about removing them? I've CC'd the known users of this repo and I've assigned it to @srwalter since he added them.

/cc @jmesmon @posborne @derekstraka @tylerwhall

jmesmon, 2016-11-02:
I've never used these, so it won't bother me if they're removed.

srwalter, 2016-11-04:
As previously discussed, I'd prefer keep these recipes.  Is there some other layout you would prefer?  I'm not partial to any particular directory for them.

cardoe, 2016-11-04:
Well they're not updated. They're dependencies and not actual target packages that install things that are consumable. I would think you are the only consumer of these since you've got some special build process to consume them because the normal `inherit cargo` packaging won't be able to use these. So it could be better for you to pull them into your own layer to continue to consume them.

cardoe, 2016-11-22:
I would also remove `rust-bin.bbclass` with this change as well.

cardoe, 2016-11-29:
If we're going to keep these then I think we need to get a simple compile test in place to ensure the build process doesn't break. Let's aim to get this resolved before pyro comes out.

cardoe, 2016-12-07:
I propose we create another layer under the meta-rust organizational umbrella and have that as the recipe layer. In theory recipes should remain mostly agnostic to the version of Yocto and it would allow for an easier place to do maintainership.

cardoe, 2018-05-30:
Alright guys. Its been 18 months. `rand-rs` hasn't pointed to a valid git commit for years. This stuff can't possibly be used by anyone? Can we remove it? If its actually used there must be other changes somewhere and I propose that this live in its own layer.

srwalter, 2018-05-30:
We actually cache all git tarballs on site so that we're immune to this sort of upstream misbehavior.  However, we are in the progress of moving more towards using cargo-bitbake, so I don't object to removing these recipes.

tylerwhall, 2018-05-31:
Agreed. We're no longer using the hand-maintained rust-bin recipes. Unfortunately we're left with no ideal solution to the space consumed by static linking, but we've been able to work with it for now.

PR #198 (2018-05-31): remove recipes-core & recipes-graphics
Remove these recipes as I do not believe they are functional since the
fixes for #126 were done (removing the modifications to the compiler to
allow them to be used). #197 additionally points out that rand-rs has
not had a valid commit hash and most of these in some way depend on that
so these cannot possibly work in their current form. Removing per #102.
Fixes #102.
commit 55e9c18a (2018-05-31): remove recipes-core & recipes-graphics

Remove these recipes as I do not believe they are functional since the
fixes for #126 were done (removing the modifications to the compiler to
allow them to be used). #197 additionally points out that rand-rs has
not had a valid commit hash and most of these in some way depend on that
so these cannot possibly work in their current form. Removing per #102.
Fixes #102.
closed 2018-05-31 by derekstraka