meta-rust/meta-rust #110: meta-rust not appendable from outside

zeenix, 2016-11-16:
I'm trying to build genivi-dev-platform here and while the main issue I encountered was version of Rust (1.7.0) GDP tracks doesn't build against gcc 6.2 I have here, I also encountered a few other issues while trying to patch it up, which seem unlikely to be specific to my distro (version) and will be filling tickets for them. Here is the first one:

The issue is that if you try to apply a patch to rust source tree from an external (to meta-rust) dir, bitbake attempts to apply the patch *before* populating the rust source tree and hence fails:

```
ERROR: rust-native-1.7.0-r0 do_patch: Command Error: 'quilt --quiltrc /home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/sysroots/x86_64-linux/etc/quiltrc push' exited with 1  Output:
Applying patch 0001-Fix-build-against-gcc-6.2.patch
can't find file to patch at input line 15
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 0f47b30e9d2670cd477ef291a4f42b83e0c70be6 Mon Sep 17 00:00:00 2001
|From: Zeeshan Ali <zeenix@gmail.com>
|Date: Tue, 15 Nov 2016 13:11:31 +0100
|Subject: [PATCH] Fix build against gcc 6.2
|
|---
| src/libbacktrace/dwarf.c |  4 ++--
| src/rt/miniz.c           | 12 ++++++++++--
| 2 files changed, 12 insertions(+), 4 deletions(-)
|
|diff --git a/src/libbacktrace/dwarf.c b/src/libbacktrace/dwarf.c
|index 55b8d7d..b5831e9 100644
|--- a/src/libbacktrace/dwarf.c
|+++ b/src/libbacktrace/dwarf.c
--------------------------
No file to patch.  Skipping patch.
2 out of 2 hunks ignored
can't find file to patch at input line 37
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/src/rt/miniz.c b/src/rt/miniz.c
|index 2b803b0..0b51856 100644
|--- a/src/rt/miniz.c
|+++ b/src/rt/miniz.c
--------------------------
No file to patch.  Skipping patch.
2 out of 2 hunks ignored
Patch 0001-Fix-build-against-gcc-6.2.patch does not apply (enforce with -f)
ERROR: rust-native-1.7.0-r0 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/work/x86_64-linux/rust-native/1.7.0-r0/temp/log.do_patch.13065

```

To reproduce against gdp, place this in meta-genivi-dev/meta-genivi-dev/recipes-devtools/rust/rust.bbappend file:

```
FILESEXTRAPATHS_append := ":${THISDIR}/${PN}"

SRC_URI_append = " \
    file://0001-Fix-build-against-gcc-6.2.patch \
"

```

And this in meta-genivi-dev/meta-genivi-dev/recipes-devtools/rust/rust/0001-Fix-build-against-gcc-6.2.patch : 

```
From 0f47b30e9d2670cd477ef291a4f42b83e0c70be6 Mon Sep 17 00:00:00 2001
From: Zeeshan Ali <zeenix@gmail.com>
Date: Tue, 15 Nov 2016 13:11:31 +0100
Subject: [PATCH] Fix build against gcc 6.2

---
 src/libbacktrace/dwarf.c |  4 ++--
 src/rt/miniz.c           | 12 ++++++++++--
 2 files changed, 12 insertions(+), 4 deletions(-)

diff --git a/src/libbacktrace/dwarf.c b/src/libbacktrace/dwarf.c
index 55b8d7d..b5831e9 100644
--- a/src/libbacktrace/dwarf.c
+++ b/src/libbacktrace/dwarf.c
@@ -2881,7 +2881,7 @@ dwarf_fileline (struct backtrace_state *state, uintptr_t pc,
     }
   else
     {
-      struct dwarf_data **pp;
+      UNUSED struct dwarf_data **pp;
 
       pp = (struct dwarf_data **) (void *) &state->fileline_data;
       while (1)
@@ -3011,7 +3011,7 @@ backtrace_dwarf_add (struct backtrace_state *state,
     {
       while (1)
 	{
-	  struct dwarf_data **pp;
+	  UNUSED struct dwarf_data **pp;
 
 	  pp = (struct dwarf_data **) (void *) &state->fileline_data;
 
diff --git a/src/rt/miniz.c b/src/rt/miniz.c
index 2b803b0..0b51856 100644
--- a/src/rt/miniz.c
+++ b/src/rt/miniz.c
@@ -575,7 +575,10 @@ tinfl_status tinfl_decompress(tinfl_decompressor *r, const mz_uint8 *pIn_buf_nex
       {
         mz_uint8 *p = r->m_tables[0].m_code_size; mz_uint i;
         r->m_table_sizes[0] = 288; r->m_table_sizes[1] = 32; TINFL_MEMSET(r->m_tables[1].m_code_size, 5, 32);
-        for ( i = 0; i <= 143; ++i) *p++ = 8; for ( ; i <= 255; ++i) *p++ = 9; for ( ; i <= 279; ++i) *p++ = 7; for ( ; i <= 287; ++i) *p++ = 8;
+        for ( i = 0; i <= 143; ++i) *p++ = 8;
+        for ( ; i <= 255; ++i) *p++ = 9;
+        for ( ; i <= 279; ++i) *p++ = 7;
+        for ( ; i <= 287; ++i) *p++ = 8;
       }
       else
       {
@@ -1393,7 +1396,12 @@ static MZ_FORCEINLINE void tdefl_find_match(tdefl_compressor *d, mz_uint lookahe
         if ((d->m_dict[probe_pos + match_len] == c0) && (d->m_dict[probe_pos + match_len - 1] == c1)) break;
       TDEFL_PROBE; TDEFL_PROBE; TDEFL_PROBE;
     }
-    if (!dist) break; p = s; q = d->m_dict + probe_pos; for (probe_len = 0; probe_len < max_match_len; probe_len++) if (*p++ != *q++) break;
+    if (!dist) break;
+    p = s;
+    q = d->m_dict + probe_pos;
+    for (probe_len = 0; probe_len < max_match_len; probe_len++)
+      if (*p++ != *q++)
+        break;
     if (probe_len > match_len)
     {
       *pMatch_dist = dist; if ((*pMatch_len = match_len = probe_len) == max_match_len) return;
-- 
2.9.3
```

Then run a clean build (important) and you should arrive at the error above. (NOTE: the patch itself is known to break the build but test is about it getting applied, not actually fixing the build).

GDP is currently tracking f13ac9d48ae928b761d7be204fa8f877d41e7099 on krogoth branch.

derekstraka, 2016-11-16:
The patch should probably be applied against `rust-source` rather than `rust`.  Does that fix your issue?

derekstraka, 2016-11-16:
NOTE: Patching `rust` with a `rust_%.bbappend` in another layer works fine on `krogoth` and `master` since the removal of shared source.

zeenix, 2016-11-16:
> The patch should probably be applied against rust-source rather than rust. Does that fix your issue?

You mean the bbappend file should be rust-source.bbappend ? I'm asking cause the patch itself is relative to the source dir. If that is what you meant, I tried that and get the same error and i can still see that source dir is not populated. There is a tmp/work-shared/rust-1.7.0-r0/rustc-1.7.0/src/ dir but it only contains an empty dir named llvm.

derekstraka, 2016-11-16:
I can look at it a little closer, but the latest on `krogoth` allows patching. Is there a technical reason to not move forward?

zeenix, 2016-11-16:
Thanks. That's good to hear, shared source is a bit complicated. The focus right now is to get a release out. Once that is done, I'd certainly want to look into not only updating to most recent krogoth at least (or even morty) but also fetching meta-rust from this canonical location.

derekstraka, 2016-11-16:
Are you pinned to krogoth for all of your other layers?

derekstraka, 2016-11-16:
I'm assume you are not and are missing 487205331bc66ad6c9e5bd0a7b2ddb5ba6b3e721.  Does it build without your bbappend?

zeenix, 2016-11-17:
> Are you pinned to krogoth for all of your other layers?

AFAICT, no. At least meta-qt5 is stuck on jethrow.

> I'm assume you are not and are missing 4872053. Does it build without your bbappend? 

On my ubuntu machine (with older gcc), yes it builds fine without the bbappend. Works fine for others too who all use ubuntu for some reason. :-P So I guess that is not the issue?

> I'll need at least your poky and meta-openembedded commits.

Sure thing but if you want to reproduce, I'll suggest you simply try building gdp: https://at.projects.genivi.org/wiki/display/GDP/GDP+Master#GDPMaster-RaspberryPi3

Here are the HEADs:

poky: 12eb72ee3b02f826a156ff4e396c770f2b93571e
meta-openembedded: 247b1267bbe95719cd4877d2d3cfbaf2a2f4865a

derekstraka, 2016-11-17:
Given the three provided HEADs, it appears to work with `rust-source.bbappend` from a layer I called `meta-test`

```
user@sparta:~/meta-rust/poky$ find meta-test/
meta-test/
meta-test/recipes-devtools
meta-test/recipes-devtools/rust
meta-test/recipes-devtools/rust/files
meta-test/recipes-devtools/rust/files/rust
meta-test/recipes-devtools/rust/files/rust/0001-Fix-build-against-gcc-6.2.patch
meta-test/recipes-devtools/rust/rust-source.bbappend
meta-test/conf
meta-test/conf/layer.conf
```
```
user@sparta:~/meta-rust/poky$ cat meta-test/recipes-devtools/rust/rust-source.bbappend
FILESEXTRAPATHS_append := ":${THISDIR}/files"

SRC_URI_append = " \
    file://rust/0001-Fix-build-against-gcc-6.2.patch \
"
```

And by working, I mean it appears to apply the patches and move forward.

zeenix, 2016-11-17:
Thanks for testing. Yeah, as I said the patch is known to break the actual build, not fix it. Please feel free to close this ticket for now then. We'll hopefully only add support for gcc6 in GDP after updating meta-rust and hopefully there is no need for a patch in there at all.

closed 2016-11-17 by zeenix