meta-rust/meta-rust #111: Rust recipe not able to clean itself

zeenix, 2016-11-16:
This one I can reproduce on Ubuntu 16.04 (gcc 5.4) host. After building GDP (for RPI3 target here) successfully (along with rust layers), I tried cleaning up only the rust layers to rebuild and it ends up with an error when I build rust afterwards:

```
$ bitbake -c cleansstate rust-llvm-native
# skipping details
NOTE: Tasks Summary: Attempted 2 tasks of which 0 didn't need to be rerun and all succeeded.
$ bitbake -c cleansstate rust
# skipping details
NOTE: Tasks Summary: Attempted 2 tasks of which 0 didn't need to be rerun and all succeeded.
$ bitbake  rust
Loading cache: 100% |#################################################################################################################################################################################################################################| ETA:  00:00:00
Loaded 2297 entries from dependency cache.
Parsing recipes: 100% |###############################################################################################################################################################################################################################| Time: 00:00:01
Parsing of 1727 .bb files complete (1726 cached, 1 parsed). 2297 targets, 324 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi3"
DISTRO            = "poky-ivi-systemd"
DISTRO_VERSION    = "11.0+snapshot-20161116"
TUNE_FEATURES     = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta              
meta-yocto        
meta-yocto-bsp    = "HEAD:12eb72ee3b02f826a156ff4e396c770f2b93571e"
meta-ivi          
meta-ivi-bsp      = "HEAD:fa4a1314e536450ad1ccfe33499b5601ac4e68c1"
meta-oe           
meta-filesystems  
meta-ruby         = "HEAD:247b1267bbe95719cd4877d2d3cfbaf2a2f4865a"
meta-qt5          = "HEAD:d715f2c1d340fa38f8a9860acc73de5e14a38b75"
meta-genivi-dev   = "HEAD:be76d217d91c8323fd2251c1c4a53d6da1a55c84"
meta-rust         = "HEAD:f13ac9d48ae928b761d7be204fa8f877d41e7099"
meta-oic          = "HEAD:69146eaf8bc05c74c377e731b7e16d82854a4659"
meta-erlang       = "HEAD:4d7eacc8e6593934ed5b0c8abc3d3e9dc339d849"
meta-rvi          = "HEAD:de9d548fe35e2cee8688faaae910b4f6f7fea17e"
meta-raspberrypi-gdp = "HEAD:be76d217d91c8323fd2251c1c4a53d6da1a55c84"
meta-raspberrypi  = "HEAD:a5f9b07a820d50ae5fb62e07306cd4e72d8638a9"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: rust-llvm-1.7.0-r0 do_compile: oe_runmake failed
ERROR: rust-llvm-1.7.0-r0 do_compile: Function failed: do_compile (log file is located at /mnt/storage/zeenix/GDP/genivi-dev-platform/gdp-src-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/rust-llvm/1.7.0-r0/temp/log.do_compile.17350)
ERROR: Logfile of failure stored in: /mnt/storage/zeenix/GDP/genivi-dev-platform/gdp-src-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/rust-llvm/1.7.0-r0/temp/log.do_compile.17350
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 12
| llvm[0]: Constructing LLVMBuild project information.
| checking for x86_64-pc-linux-gnu-clang... gcc
| checking for C compiler default output file name... a.out
| checking whether the C compiler works... yes
| checking whether we are cross compiling... no
| checking for suffix of executables...
| checking for suffix of object files... o
| checking whether we are using the GNU C compiler... yes
| checking whether gcc accepts -g... yes
| checking for gcc option to accept ISO C89... none needed
| checking whether we are using the GNU C++ compiler... yes
| checking whether g++ accepts -g... yes
| checking how to run the C preprocessor... arm-poky-linux-gnueabi-gcc -E --sysroot=/mnt/storage/zeenix/GDP/genivi-dev-platform/gdp-src-build/tmp/sysroots/raspberrypi3  -march=armv7ve -marm -mfpu=neon-vfpv4  -mfloat-abi=hard -mcpu=cortex-a7
| checking whether GCC or Clang is our host compiler... gcc
| checking build system type... x86_64-pc-linux-gnu
| checking host system type... x86_64-pc-linux-gnu
| checking target system type... x86_64-pc-linux-gnu
[... 456 lines cut ...]

On Fedora 25 (with gcc 6.2), I had a different outcome but there I build didn't succeed the first time so can not be sure the symptoms on F25 are related to this bug or not:

```
WARNING: rust-llvm-native-1.7.0-r0 do_populate_lic: Could not copy license file /home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/work-shared/rust-1.7.0-r0/rustc-1.7.0/src/llvm/LICENSE.TXT to /home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.7.0-r0/license-destdir/rust-llvm-native/LICENSE.TXT: [Errno 2] No such file or directory: '/home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/work-shared/rust-1.7.0-r0/rustc-1.7.0/src/llvm/LICENSE.TXT'
ERROR: rust-llvm-native-1.7.0-r0 do_install: oe_runmake failed
ERROR: rust-llvm-native-1.7.0-r0 do_install: Function failed: do_install (log file is located at /home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.7.0-r0/temp/log.do_install.11173)
ERROR: Logfile of failure stored in: /home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.7.0-r0/temp/log.do_install.11173
Log data follows:
| DEBUG: Executing shell function do_install
| NOTE: make -j 8 DESTDIR=/home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.7.0-r0/image install
| make: *** No rule to make target 'install'.  Stop.
| ERROR: oe_runmake failed
| ERROR: Function failed: do_install (log file is located at /home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.7.0-r0/temp/log.do_install.11173)
ERROR: Task 2 (virtual:native:/home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/../meta-rust/recipes-devtools/rust/rust-llvm.bb, do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 59 tasks of which 52 didn't need to be rerun and 1 failed.
No currently running tasks (59 of 61)

Summary: 1 task failed:
  virtual:native:/home/zeenix/checkout/GDP/genivi-dev-platform/gdp-src-build/../meta-rust/recipes-devtools/rust/rust-llvm.bb, do_install
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

Isn't this the same as reported in this comment? https://at.projects.genivi.org/jira/browse/GDP-375?focusedCommentId=14056&page=com.atlassian.jira.plugin.system.issuetabpanels:comment-tabpanel#comment-14056 and AFAIK Robert Marshall uses Ubuntu, not Fedora.

derekstraka, 2016-11-16:
What are you intending to rebuild with your clean?  Are you trying to rebuild the rust cross compiler?  It looks like it is trying to build a compiler to run on the target since it is building `rust-llvm` for the target not native.

zeenix, 2016-11-17:
Glad you asked cause if i build the target image instead, everything works fine. I was expecting 'rust' recipe to be a dep of the image and hence just told bitbake to build 'rust' to save time. I should have been more specific and did `bitbake rust-llvm-native`.

closed 2016-11-17 by zeenix