meta-rust/meta-rust #113: libstd-rs fails with "libcompiler-rt.a: ...... recompile with -fPIC" when building for MIPS

Fulkerson, 2016-11-17:
I'm trying to make meta-rust work with a mips32r2 little endian target. Things are progressing nicely and I've managed to build rust properly.

However the libstd-rs recipe failes with the following error:

    ....../usr/lib/libcompiler-rt.a(mulodi4.o): relocation R_MIPS_26 against `__divdi3@@GLIBC_2.0' can not be used when making a shared object; recompile with -fPIC

I did as told and added `-fPIC` to `COMMON_INCLUDES=` in _compiler-rt/make/config.mk_ and now things work well enough. I am able to build and run the rust-hello-world recipe on target.

Does anyone have an idea why -fPIC is not added by default. Should it? If it should, what is the best practice? I don't think COMMON_INCLUDES is the proper place after all.

Fulkerson, 2016-11-17:
Did some more digging. It seems as though -fPIC was removed from compiler-rt in commit a4b1f78283db06dc5bbec6abaab7a93086697c44.

"Rust: Silence -fPIC warnings on Windows"

However it is back again in at least version 1.12.1 of rust, so I think this will solve itself in future versions of rust.

cardoe, 2016-11-21:
@Fulkerson Are you using master or one of the other branches? Do you have any proposed patches?

Fulkerson, 2016-11-27:
I am using master. My solution was to simply revert commit a4b1f78283db06dc5bbec6abaab7a93086697c44 of rust. However it is already fixed upstream so when rust is bumped it should fix itself.

meta-rust does not currently work for MIPS anyway. I plan on making a pull request when stuff seem to work properly!

jmesmon, 2016-12-06:
morty & master (pyro) now ship 1.12.1

derekstraka, 2017-01-30:
@Fulkerson - Do you have a branch where the changes for MIPS live?  I can touch things up for 1.14 if you have a starting point and add it to the CI build.

closed 2017-03-12 by Fulkerson