meta-rust/meta-rust #126: crasher when using -C crate_hash

cardoe, 2016-12-05:
## Version(s) of meta-rust

- master / Rust 1.12.1
- krogoth-1.12.1 / Rust 1.12.1

## Version(s) of poky and/or oe-core

- krogoth
- morty

## Steps to reproduce

Hard to reproduce but I've made it happen with two packages. At one point it was happening in this repo with which prompted me to remove `-C crate_hash` from the build of cargo itself.

cardoe, 2016-12-05:
The crash:

```
| error: internal compiler error: /var/build/build/tmp/work/x86_64-linux/rust-native/1.12.1-r0/rustc-1.12.1/src/librustc_metadata/decoder.rs:74: lookup_item: id not found: DefIndex(101) in crate "uuid" with number 23
|
| note: the compiler unexpectedly panicked. this is a bug.
|
| note: we would appreciate a bug report: https://github.com/rust-lang/rust/blob/master/CONTRIBUTING.md#bug-reports
|
| note: run with `RUST_BACKTRACE=1` for a backtrace
|
| thread 'rustc' panicked at 'Box<Any>', /var/build/build/tmp/work/x86_64-linux/rust-native/1.12.1-r0/rustc-1.12.1/src/librustc_errors/lib.rs:634
| stack backtrace:
|    1:     0x7fd5edbbff69 - <unknown>
|    2:     0x7fd5edbd07ac - <unknown>
|    3:     0x7fd5edbceba4 - <unknown>
|    4:     0x7fd5edbcf308 - std::panicking::rust_panic_with_hook::h8d486474663979b9
|    5:     0x7fd5e6d337f7 - <unknown>
|    6:     0x7fd5e6d44325 - rustc_errors::Handler::bug::h126e43857fe80736
|    7:     0x7fd5eae30194 - <unknown>
|    8:     0x7fd5ead707d5 - <unknown>
|    9:     0x7fd5ead70612 - rustc::session::bug_fmt::hc0d300eca0d34bd4
|   10:     0x7fd5eb496aa2 - rustc_metadata::decoder::get_adt_def::h2b5e55915402268b
|   11:     0x7fd5eb4b53b4 - rustc_metadata::csearch::_<impl rustc..middle..cstore..CrateStore<'tcx> for rustc_metadata..cstore..CStore>::adt_def::h8d598db111499922
|   12:     0x7fd5eae11ce7 - rustc::ty::_<impl rustc..ty..context..TyCtxt<'a, 'gcx, 'tcx>>::lookup_adt_def_master::h9f53e611a86e180f
|   13:     0x7fd5eae121f5 - rustc::ty::_<impl rustc..ty..context..TyCtxt<'a, 'gcx, 'tcx>>::lookup_adt_def::ha278cd2d0e5dd00e
|   14:     0x7fd5eb476119 - rustc_metadata::tydecode::TyDecoder::parse_ty::hcfebf7f7d8f63b14
|   15:     0x7fd5eb4759e0 - rustc_metadata::tydecode::TyDecoder::parse_ty::hcfebf7f7d8f63b14
|   16:     0x7fd5eb4975cd - rustc_metadata::decoder::get_type::h4ab79813499084a2
|   17:     0x7fd5eb4966ef - rustc_metadata::decoder::get_adt_def::h2b5e55915402268b
|   18:     0x7fd5eb4b53b4 - rustc_metadata::csearch::_<impl rustc..middle..cstore..CrateStore<'tcx> for rustc_metadata..cstore..CStore>::adt_def::h8d598db111499922
|   19:     0x7fd5eae11ce7 - rustc::ty::_<impl rustc..ty..context..TyCtxt<'a, 'gcx, 'tcx>>::lookup_adt_def_master::h9f53e611a86e180f
|   20:     0x7fd5eae121f5 - rustc::ty::_<impl rustc..ty..context..TyCtxt<'a, 'gcx, 'tcx>>::lookup_adt_def::ha278cd2d0e5dd00e
|   21:     0x7fd5eb476119 - rustc_metadata::tydecode::TyDecoder::parse_ty::hcfebf7f7d8f63b14
|   22:     0x7fd5eb4759e0 - rustc_metadata::tydecode::TyDecoder::parse_ty::hcfebf7f7d8f63b14
|   23:     0x7fd5eb4730e7 - rustc_metadata::tydecode::TyDecoder::parse_substs::h0a73406b85b85a04
|   24:     0x7fd5eb4996d8 - rustc_metadata::decoder::get_impl_trait::h45af60a7d28fff8c
|   25:     0x7fd5eb4b62af - rustc_metadata::csearch::_<impl rustc..middle..cstore..CrateStore<'tcx> for rustc_metadata..cstore..CStore>::impl_trait_ref::h530d3f922fabda73
|   26:     0x7fd5eae0eb91 - rustc::ty::_<impl rustc..ty..context..TyCtxt<'a, 'gcx, 'tcx>>::impl_trait_ref::hb8cee49d6f7024ff
|   27:     0x7fd5eae15a7e - rustc::ty::_<impl rustc..ty..context..TyCtxt<'a, 'gcx, 'tcx>>::populate_implementations_for_trait_if_necessary::hcf8010d4c0e15ec3
|   28:     0x7fd5eae14686 - rustc::ty::_<impl rustc..ty..context..TyCtxt<'a, 'gcx, 'tcx>>::trait_has_default_impl::h2aabf2bdf819852e
|   29:     0x7fd5ec282e25 - _<rustc_typeck..coherence..orphan..OrphanChecker<'cx, 'tcx> as rustc..hir..intravisit..Visitor<'v>>::visit_item::h8497006e991b3a06
|   30:     0x7fd5ec28b045 - rustc_typeck::coherence::check_coherence::hd035dc985060f8ee
|   31:     0x7fd5ec2939f3 - rustc_typeck::check_crate::hc2a0229872cc6074
|   32:     0x7fd5edf75e36 - <unknown>
|   33:     0x7fd5edec1bb8 - <unknown>
|   34:     0x7fd5edf2a1f8 - rustc_driver::driver::compile_input::h8f39049cdedc4958
|   35:     0x7fd5edf56f16 - rustc_driver::run_compiler::hdb93d6e9edbdb1b8
|   36:     0x7fd5ede9e122 - <unknown>
|   37:     0x7fd5edbdc9e6 - __rust_maybe_catch_panic
|   38:     0x7fd5edeb77da - <unknown>
|   39:     0x7fd5edbccf02 - <unknown>
|   40:     0x7fd5e64d9183 - start_thread
|   41:     0x7fd5ed80a37c - clone
|   42:                0x0 - <unknown>
```

jmesmon, 2016-12-05:
One of my nagging suspicions about our use of `-C crate_hash=${BB_TASKHASH}` in `RUSTFLAGS` is that it doesn't necessarily appear to handle the case of 2 versions of the same-named crate being used within a single recipe (and given cargo's large number of dependencies, I wouldn't be surprised if it had 2 versions of the same crate included).

Not clear to me why/how that would trigger this compiler panic, though. Maybe some machinery that assumes each crate has a unique `LinkMeta` and blindly ignores some of the data for the earlier or later added crate?

cardoe, 2016-12-05:
Its falling down here: https://github.com/rust-lang/rust/blob/1.12.1/src/librustc_metadata/decoder.rs#L74

cardoe, 2016-12-05:
Yeah both of the projects where this fails are pulling in `uuid` 0.3.1 and another dependency pulls in `uuid` 0.2.3. And looking through the failed commits in #116, I was originally pulling in `term` 0.2.14 and `term` 0.4.4 to compile Cargo but got rid of both in a later commit by moving another dependency to a newer one that pulled in `term` 0.4.4.

cardoe, 2016-12-05:
So I'm of the opinion of reverting that patch out. Correctness > Caching. My understanding is that this change is for purely caching.

cardoe, 2016-12-05:
The patch I'm suggesting we remove is https://github.com/meta-rust/meta-rust/blob/master/recipes-devtools/rust/files/rust-1.12.1/0007-Allow-overriding-crate_hash-with-C-crate_hash.patch

cardoe, 2016-12-05:
If I understand this correctly, this hash is just when we're building against .so's and it doesn't matter if you're building a Cargo project that's not using shared objects. So the answer might be as simple as not passing this flag on to Cargo built projects?

cardoe, 2016-12-05:
/cc @meta-rust/maintainers Might be worth having others weigh in on changes that might affect your projects.

derekstraka, 2016-12-05:
I would prefer it to be removed and be slower without the chance of ICEs.  It looks like the patch will need to be reworked in the near future anyway even if we do nothing (https://github.com/rust-lang/rust/pull/35854/files).

jmesmon, 2016-12-05:
Alternate to removing entirely is pushing it further down the call chain for crate hash generation so we pick up enough of the hash to include the version number, but avoid the potentially non-deterministic bits.

I'm fine with removal though, as the above is likely difficult to confirm we've got right.

IIRC, this was originally added as @srwalter was seeing bitbake complain about sstate mismatches. I was never entirely clear on what caused bitbake to complain, though.

jmesmon, 2016-12-06:
Original addition: https://github.com/meta-rust/meta-rust/pull/33

jmesmon, 2016-12-06:
Another alternate is just to confine it's use to the `rust-bin`  based recipes, where it should be safe by virtue of how those recipes work.

cardoe, 2016-12-06:
Right. I don't see how this is useful for cargo based builds that are building a binary. Cargo will pull in all the dependencies itself and link them all together. This is only really useful for building each crate individually and installing it into the system ala `rust-bin` and then when you compile the library it reads those rlibs in to link them together.

cardoe, 2016-12-06:
Its worth noting that it will still fail for things that require multiple versions of the same crate.