meta-rust/meta-rust #130: Building a Toolchain Installer Fails

leon-anavi, 2016-12-23:
## Version(s) of meta-rust

Krogoth

## Version(s) of poky and/or oe-core

Krogoth

## Expected result

To successfully build a toolchain installer using bitbake <image> -c populate_sdk.

## Actual result

The following errors appear:

```
ERROR: core-image-minimal-1.0-r0 do_populate_sdk: Unable to install packages. Command '/media/storage/poky-rust/build/tmp/sysroots/x86_64-linux/usr/bin/smart --log-level=warning --data-dir=/media/storage/poky-rust/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-minimal/1.0-r0/sdk/image/opt/poky/2.1.2/sysroots/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/var/lib/smart install --attempt -y base-passwd-dev@cortexa7hf_neon_vfpv4 glibc-locale-dbg@cortexa7hf_neon_vfpv4 run-postinsts-dbg@all initscripts-dbg@cortexa7hf_neon_vfpv4 base-files-dev@raspberrypi3 sysvinit-dbg@cortexa7hf_neon_vfpv4 sysvinit-inittab-dev@raspberrypi3 modutils-initscripts-dbg@cortexa7hf_neon_vfpv4 keymaps-dev@raspberrypi3 busybox-dbg@cortexa7hf_neon_vfpv4 kbd-dbg@cortexa7hf_neon_vfpv4 eudev-dbg@cortexa7hf_neon_vfpv4 run-postinsts-dev@all gcc-runtime-dbg@cortexa7hf_neon_vfpv4 packagegroup-core-boot-dbg@raspberrypi3 util-linux-dev@cortexa7hf_neon_vfpv4 init-ifupdown-dbg@cortexa7hf_neon_vfpv4 libz-dev@cortexa7hf_neon_vfpv4 netbase-dev@cortexa7hf_neon_vfpv4 busybox-dev@cortexa7hf_neon_vfpv4 rust-hello-world-dbg@cortexa7hf_neon_vfpv4 sysvinit-inittab-dbg@raspberrypi3 base-passwd-dbg@cortexa7hf_neon_vfpv4 packagegroup-core-standalone-sdk-target-dev@all update-rc.d-dev@all sysvinit-dev@cortexa7hf_neon_vfpv4 kbd-dev@cortexa7hf_neon_vfpv4 keymaps-dbg@raspberrypi3 kmod-dbg@cortexa7hf_neon_vfpv4 modutils-initscripts-dev@cortexa7hf_neon_vfpv4 update-rc.d-dbg@all opkg-utils-dbg@cortexa7hf_neon_vfpv4 eudev-dev@cortexa7hf_neon_vfpv4 base-files-dbg@raspberrypi3 initscripts-dev@cortexa7hf_neon_vfpv4 util-linux-dbg@cortexa7hf_neon_vfpv4 packagegroup-core-boot-dev@raspberrypi3 libz-dbg@cortexa7hf_neon_vfpv4 kmod-dev@cortexa7hf_neon_vfpv4 rust-hello-world-dev@cortexa7hf_neon_vfpv4 netbase-dbg@cortexa7hf_neon_vfpv4 init-ifupdown-dev@cortexa7hf_neon_vfpv4 opkg-utils-dev@cortexa7hf_neon_vfpv4' returned 1:
Loading cache...
Updating cache...               ######################################## [100%]

Computing transaction...warning: Can't install rust-hello-world-dev-git-r0@cortexa7hf_neon_vfpv4: Can't install compiler-rt-dev-1.12.1-r0@cortexa7hf_neon_vfpv4: no package provides compiler-rt = 1.12.1-r0
```

`
error: compiler-rt-dev-1.12.1-r0 conflicts with compiler-rt = 1.12.1-r0
`

## Steps to reproduce
* Clone Poky
* Clone meta-raspberrypi
* Clone meta-rust
* Initialize the environment
* Set MACHINE to raspberrypi3
* Add rust-hello-world to the image but appending the following line to conf/local.conf
`
IMAGE_INSTALL_append = " rust-hello-world"
`
* Build a toolchain Installer:
`
bitbake core-image-minimal -c populate_sdk
`

Is this a known issue and is there a fix for it?

Best regards,
Leon

derekstraka, 2016-12-23:
AFAIK, no one has ever validated this before.  I assume you want to be able to distribute the rust toolchain as part of an SDK build?

cardoe, 2016-12-23:
This is on the README as a TODO to be implemented.

leon-anavi, 2016-12-23:
Hi @derekstraka, @cardoe,

@derekstraka, yes. Actually this was reported in GENIVI Development Platform because the RVI SOTA client relies on meta-rust. I simplified the case and reproduce it to confirm that the issue is meta-rust specific. Details are available at: https://at.projects.genivi.org/jira/browse/GDP-539

@cardoe, ok, noted. Thank you for the provided information.

Thanks,
Leon

cardoe, 2016-12-23:
@leon-anavi I've never used the SDK bits of Yocto and as a result I'm not sure how to properly setup recipes to work. If you have some suggested changes and some docs that you can point me to I'll happily make those changes to this layer.

leon-anavi, 2016-12-23:
@cardoe I am also not an expert about the SDK bits of Yocto. After several experiments the only way that I found to fix the issue was to ensure that packages -dev and -devstatic remain empty. This way there is no duplication of libcompiler-rt.a in compiler-rt and compiler-rt-dev. As a result populate_sdk now works. Please have a look at my patch: https://github.com/meta-rust/meta-rust/pull/131

Best regards,
Leon

cardoe, 2016-12-26:
Fixed by #131.

commit 2c923bbd (2016-12-26): Merge pull request #131 from leon-anavi/krogoth

compiler-rt: fix toolchain / sdk installer

Ensures files are placed in the proper sub-packages. fixes #130.
closed 2016-12-26 by cardoe