meta-rust/meta-rust #134: Building Cargo for ARM Fails

leon-anavi, 2016-12-29:
## Version(s) of meta-rust

Krogoth

## Version(s) of poky and/or oe-core

Krogoth

## Expected result

The expected result is to successfully built cargo.

## Actual result

```
ERROR: cargo-0.13.0-r0 do_compile: oe_runmake failed
ERROR: cargo-0.13.0-r0 do_compile: Function failed: do_compile (log file is located at /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/temp/log.do_compile.30503)
ERROR: Logfile of failure stored in: /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/temp/log.do_compile.30503
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 VERBOSE=1
| /media/storage/poky-rust/build/tmp/sysroots/x86_64-linux/usr/bin/rustc -V
| rustc 1.12.1
| target/snapshot/bin/cargo --version
| cargo 0.13.0-nightly (e713e7f 2016-08-31)
| target/snapshot/bin/cargo build --target arm-poky-linux-gnueabi --manifest-path /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git//Cargo.toml --release  --verbose
|    Compiling regex-syntax v0.3.4
|    Compiling crossbeam v0.2.9
|    Compiling winapi v0.2.8
|    Compiling pkg-config v0.3.8
|    Compiling winapi-build v0.1.1
|    Compiling strsim v0.3.0
|    Compiling glob v0.2.11
|    Compiling rustc-serialize v0.3.19
|      Running `rustc /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/cargo_home/bitbake/regex-syntax-0.3.4/src/lib.rs --crate-name regex_syntax --crate-type lib -C opt-level=3 -C metadata=5f894c94a1e04549 -C extra-filename=-5f894c94a1e04549 --out-dir /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --emit=dep-info,link --target arm-poky-linux-gnueabi -C linker=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/wrapper/target-rust-ccld -L dependency=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --cap-lints allow`
|      Running `rustc /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/cargo_home/bitbake/crossbeam-0.2.9/src/lib.rs --crate-name crossbeam --crate-type lib -C opt-level=3 -C metadata=c6acd65498c68535 -C extra-filename=-c6acd65498c68535 --out-dir /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --emit=dep-info,link --target arm-poky-linux-gnueabi -C linker=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/wrapper/target-rust-ccld -L dependency=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --cap-lints allow`
|      Running `rustc /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/cargo_home/bitbake/winapi-0.2.8/src/lib.rs --crate-name winapi --crate-type lib -C opt-level=3 -C metadata=d29f7fa53c1342e6 -C extra-filename=-d29f7fa53c1342e6 --out-dir /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --emit=dep-info,link --target arm-poky-linux-gnueabi -C linker=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/wrapper/target-rust-ccld -L dependency=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --cap-lints allow`
|      Running `rustc /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/cargo_home/bitbake/pkg-config-0.3.8/src/lib.rs --crate-name pkg_config --crate-type lib -C opt-level=3 -C metadata=644ff8561a5494b1 -C extra-filename=-644ff8561a5494b1 --out-dir /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/release/deps --emit=dep-info,link -C linker=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/wrapper/build-rust-ccld -L dependency=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/release/deps --cap-lints allow`
|      Running `rustc /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/cargo_home/bitbake/winapi-build-0.1.1/src/lib.rs --crate-name build --crate-type lib -C opt-level=3 -C metadata=812169c879249320 -C extra-filename=-812169c879249320 --out-dir /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/release/deps --emit=dep-info,link -C linker=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/wrapper/build-rust-ccld -L dependency=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/release/deps --cap-lints allow`
|      Running `rustc /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/cargo_home/bitbake/strsim-0.3.0/src/lib.rs --crate-name strsim --crate-type lib -C opt-level=3 -C metadata=4dbe3971de3d56d0 -C extra-filename=-4dbe3971de3d56d0 --out-dir /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --emit=dep-info,link --target arm-poky-linux-gnueabi -C linker=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/wrapper/target-rust-ccld -L dependency=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --cap-lints allow`
|      Running `rustc /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/cargo_home/bitbake/glob-0.2.11/src/lib.rs --crate-name glob --crate-type lib -C opt-level=3 -C metadata=4255f5b8721aa746 -C extra-filename=-4255f5b8721aa746 --out-dir /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --emit=dep-info,link --target arm-poky-linux-gnueabi -C linker=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/wrapper/target-rust-ccld -L dependency=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --cap-lints allow`
|      Running `rustc /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/cargo_home/bitbake/rustc-serialize-0.3.19/src/lib.rs --crate-name rustc_serialize --crate-type lib -C opt-level=3 -C metadata=3fe92c01c925aa1b -C extra-filename=-3fe92c01c925aa1b --out-dir /media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --emit=dep-info,link --target arm-poky-linux-gnueabi -C linker=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/wrapper/target-rust-ccld -L dependency=/media/storage/poky-rust/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cargo/0.13.0-r0/git/target/arm-poky-linux-gnueabi/release/deps --cap-lints allow`
| error[E0463]: can't find crate for `std`
| 
| error: aborting due to previous error
| 
| Build failed, waiting for other jobs to finish...
| error[E0463]: can't find crate for `std`
| 
| error: aborting due to previous error
| 
| error[E0463]: can't find crate for `std`
| 
| error: aborting due to previous error
| 
| Build failed, waiting for other jobs to finish...
| Build failed, waiting for other jobs to finish...
| error[E0463]: can't find crate for `std`
| 
| error: aborting due to previous error
| 
| Build failed, waiting for other jobs to finish...
| error[E0463]: can't find crate for `std`
| 
| error: aborting due to previous error
| 
| Build failed, waiting for other jobs to finish...
| error[E0463]: can't find crate for `std`
| 
| error: aborting due to previous error
[... 13 lines cut ...]

Summary: 1 task failed:
  /media/storage/poky-rust/meta-rust/recipes-devtools/cargo/cargo_0.13.0.bb, do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

## Steps to reproduce

* Clone Poky
* Clone meta-raspberrypi
* Clone meta-rust
* Initialize the environment
* Set MACHINE to raspberrypi3
* Add rust-hello-world to the image but appending the following line to conf/local.conf
`
IMAGE_INSTALL_append = " rust-hello-world"
`
* Build a toolchain Installer:
`
bitbake cargo
`

Best regards,
Leon

derekstraka, 2016-12-29:
Data point - The nightly with qemu-arm worked for cargofmt (https://ci.asterius.io/job/meta-rust-nightly/job/krogoth/7/flowGraphTable/).

derekstraka, 2016-12-29:
I was wrongly assuming this was cargo-native and not target cargo.  IIRC, this is a known deficiency that you cannot create a cargo/rustc that will execute on the target.  @cardoe

cardoe, 2016-12-29:
Well good to know that its building Rust for the target. I didn't think that worked. This is really issue #81.

cardoe, 2016-12-29:
@leon-anavi Is your goal to be able to run the Rust compiler and Cargo on the actual target? Cause you're not building a toolchain afaik here. Toolchains are the SDK packages afaik.

derekstraka, 2017-01-30:
@leon-anavi - anything to add on the use case; otherwise, I will assume this is a duplicate of #81

leon-anavi, 2017-01-30:
I don't have any additional comments at the moment regarding this issue.

Best regards,
Leon

derekstraka, 2017-01-30:
Duplicate of #81

closed 2017-01-30 by derekstraka