meta-rust/meta-rust #136: unable to create git based recipes with the crate fetcher

cardoe, 2017-01-01:
Using the crate fetcher will fail with git based recipes that use SRCPV in their version information as recommended by the Yocto documentation. This is because BitBake loads the fetcher after resolving SRCPV. I've opened an issue upstream. Since this fetcher is used in our krogoth, morty and pyro branches it will affect all 3.

https://bugzilla.yoctoproject.org/show_bug.cgi?id=10867

cardoe, 2017-01-01:
Users will see errors like this:

```
ERROR: ExpansionError during parsing /var/build/poky/meta-rust/recipes-example/rustfmt/rustfmt_git.bb
Traceback (most recent call last):
  File "/var/build/poky/bitbake/lib/bb/data_smart.py", line 412, in DataSmart.expandWithRefs(s="${@bb.parse.BBHandler.vars_from_file(d.getVar('FILE', False),d)[1] or '1.0'}.${SRCPV}", varname='PV'):
                 try:
    >                s = __expand_var_regexp__.sub(varparse.var_sub, s)
                     try:
  File "/var/build/poky/bitbake/lib/bb/data_smart.py", line 111, in VariableParse.var_sub(match=<_sre.SRE_Match object; span=(77, 85), match='${SRCPV}'>):
                 else:
    >                var = self.d.getVarFlag(key, "_content")
                 self.references.add(key)
  File "/var/build/poky/bitbake/lib/bb/data_smart.py", line 773, in DataSmart.getVarFlag(var='SRCPV', flag='_content', expand=True, noweakdefault=False, parsing=False):
                     cachename = var + "[" + flag + "]"
    >            value = self.expand(value, cachename)

  File "/var/build/poky/bitbake/lib/bb/data_smart.py", line 436, in DataSmart.expand(s='${@bb.fetch2.get_srcrev(d)}', varname='SRCPV'):
         def expand(self, s, varname = None):
    >        return self.expandWithRefs(s, varname).value

  File "/var/build/poky/bitbake/lib/bb/data_smart.py", line 426, in DataSmart.expandWithRefs(s='${@bb.fetch2.get_srcrev(d)}', varname='SRCPV'):
                 except Exception as exc:
    >                raise ExpansionError(varname, s, exc) from exc

bb.data_smart.ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception NoMethodError: Could not find a fetcher which supports the URL: 'crate://crates.io/aho-corasick/0.5.3'
```

hmvp, 2018-04-20:
This is also an issue if you try to use the devtool tool to modify recipes. It tries to do some parsing for versions etc and borks on the same issue

cardoe, 2018-05-22:
I'm all for merging the crate fetcher upstream.

Progdrasil, 2019-10-24:
I'm still having this issue using `devtool` on the `thud` branch

alex88, 2021-05-17:
Is there a workaround on this for now?

russell-islam, 2022-02-10:
Any plan to address this issue?