meta-rust/meta-rust #138: use RUST_TARGET_PATH instead of patching rustc

cardoe, 2017-01-03:
Instead of patching rustc to load the proper target for us. We should use the environment variable to ensure that the compiler gets the right target. This will lead to more consistent data when using shared SSTATE between different machines.

see japaric/xargo#44 and rust-lang/rust#24666

jmesmon, 2017-01-03:
The existing patch to rustc for locating targets was really designed just to add a default value to RUST_TARGET_PATH. Should be possible to simply drop the patch & always set RUST_TARGET_PATH to the path that would have been patched in.

But based on the mention of the sstate stuff, I'm guessing you want to use multiple RUST_TARGET_PATH elements to allow the yocto-target libstd & target.json to move to a location outside of the yocto-host sysroot (which still having the host libstd & target.json in the host sysroot)? Or something along those lines?

cardoe, 2017-01-03:
I was more thinking of your first paragraph. I hadn't really thought about moving things out of the host sysroot yet.

cardoe, 2017-01-03:
I was trying to use rustbuild and ran into something where the path was getting baked in when `--target` was a full path. I know that the `crate_hash` bits exist to try and neutralize some of the sstate changes due to paths so I thought this might help all around.

jmesmon, 2017-01-03:
We're not currently passing `--target` a full path, only `${HOST_SYS}` (as far as I can see). Even for the bootstrapping (where we're already using `RUST_TARGET_PATH`).

Is there something fiddly about `rustbuild` that is causing a full path to be passed?

cardoe, 2017-09-28:
This is now finished.

closed 2017-09-28 by cardoe