meta-rust/meta-rust #143: Use static versions of downloaded tarballs

DunderRoffe, 2017-01-30:
Hi,
I work as maintainer for [genivi-dev-platform](https://github.com/GENIVI/genivi-dev-platform) which uses meta-rust. Sometimes the builds start to fail without any layer being updated due to compilation errors in meta-rust. Therefore, it would be nice if you started to use specific versions for the tarballs that are downloaded install time.

## Version(s) of meta-rust
2879fc131d8ac1928720f029887bbc0401234119

## Version(s) of poky and/or oe-core
poky: 12eb72ee3b02f826a156ff4e396c770f2b93571e

## Expected result
Repeated builds should always reproduce the same result

## Actual result
Compilation may fail whenever llvm-rust is updated without bumping the version in the repo.

cardoe, 2017-01-30:
This doesn't make any sense. We are using a rust-llvm fixed source from within the rust source tarball. We are not dynamically fetching anything down. Please be more clear in the issue you are having.

cardoe, 2017-01-30:
You referenced the script that I use to pull down each commit in the OE's krogoth stable series to ensure we still work.

derekstraka, 2017-01-30:
Do you have any logs of the actual build failures?

derekstraka, 2017-01-30:
Is your build node running out of memory compiling LLVM?

DunderRoffe, 2017-01-31:
Maybe I have misunderstood something. Will double check that there were no other source code change that effected this.

Anyway here is the error log I get:
```
ERROR: rust-llvm-native-1.12.1-r0 do_compile: oe_runmake failed
ERROR: rust-llvm-native-1.12.1-r0 do_compile: Function failed: do_compile (log file is located at /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/temp/log.do_compile.30063)
ERROR: Logfile of failure stored in: /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/temp/log.do_compile.30063
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 12
| /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/sysroots/x86_64-linux/usr/bin/cmake -H/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/rustc-1.12.1/src/llvm -B/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build --check-build-system CMakeFiles/Makefile.cmake 0
| /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/sysroots/x86_64-linux/usr/bin/cmake -E cmake_progress_start /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build/CMakeFiles /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build/CMakeFiles/progress.marks
| make -f CMakeFiles/Makefile2 all
| make[1]: Entering directory '/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build'
| make -f lib/Support/CMakeFiles/LLVMSupport.dir/build.make lib/Support/CMakeFiles/LLVMSupport.dir/depend
| make -f lib/TableGen/CMakeFiles/LLVMTableGen.dir/build.make lib/TableGen/CMakeFiles/LLVMTableGen.dir/depend
| make -f utils/TableGen/CMakeFiles/obj.llvm-tblgen.dir/build.make utils/TableGen/CMakeFiles/obj.llvm-tblgen.dir/depend
| make -f CMakeFiles/CONFIGURE_LLVM_NATIVE.dir/build.make CMakeFiles/CONFIGURE_LLVM_NATIVE.dir/depend
| make -f lib/Transforms/Hello/CMakeFiles/LLVMHello_exports.dir/build.make lib/Transforms/Hello/CMakeFiles/LLVMHello_exports.dir/depend
| make -f lib/MC/CMakeFiles/LLVMMC.dir/build.make lib/MC/CMakeFiles/LLVMMC.dir/depend
| make -f lib/MC/MCParser/CMakeFiles/LLVMMCParser.dir/build.make lib/MC/MCParser/CMakeFiles/LLVMMCParser.dir/depend
| make -f lib/MC/MCDisassembler/CMakeFiles/LLVMMCDisassembler.dir/build.make lib/MC/MCDisassembler/CMakeFiles/LLVMMCDisassembler.dir/depend
| make -f lib/ObjectYAML/CMakeFiles/LLVMObjectYAML.dir/build.make lib/ObjectYAML/CMakeFiles/LLVMObjectYAML.dir/depend
| make -f lib/Option/CMakeFiles/LLVMOption.dir/build.make lib/Option/CMakeFiles/LLVMOption.dir/depend
| make -f lib/DebugInfo/CodeView/CMakeFiles/LLVMDebugInfoCodeView.dir/build.make lib/DebugInfo/CodeView/CMakeFiles/LLVMDebugInfoCodeView.dir/depend
| make -f lib/DebugInfo/DWARF/CMakeFiles/LLVMDebugInfoDWARF.dir/build.make lib/DebugInfo/DWARF/CMakeFiles/LLVMDebugInfoDWARF.dir/depend
| make[2]: Entering directory '/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build'
| cd /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build && /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/sysroots/x86_64-linux/usr/bin/cmake -E cmake_depends "Unix Makefiles" /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/rustc-1.12.1/src/llvm /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/rustc-1.12.1/src/llvm /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build/CMakeFiles/CONFIGURE_LLVM_NATIVE.dir/DependInfo.cmake --color=
| make[2]: Leaving directory '/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build'
| make -f CMakeFiles/CONFIGURE_LLVM_NATIVE.dir/build.make CMakeFiles/CONFIGURE_LLVM_NATIVE.dir/build
| make[2]: Entering directory '/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build'
| [  0%] Configuring NATIVE LLVM...
| cd /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build/NATIVE && /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/sysroots/x86_64-linux/usr/bin/cmake -G Unix\ Makefiles /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/rustc-1.12.1/src/llvm -DLLVM_TARGET_IS_CROSSCOMPILE_HOST=TRUE
| make[2]: Entering directory '/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build'
| cd /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build && /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/sysroots/x86_64-linux/usr/bin/cma [... cut ...]
| make[1]: Leaving directory '/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/build'
| Makefile:152: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/tmp/work/x86_64-linux/rust-llvm-native/1.12.1-r0/temp/log.do_compile.30063)
ERROR: Task 5007 (virtual:native:/mnt/storage/visj/genivi-dev-platform-minnowboard/gdp-src-build/../meta-rust/recipes-devtools/rust/rust-llvm_1.12.1.bb, do_compile) failed with exit code '1'
```

srwalter, 2017-01-31:
Oh!  I've seen that error before.  That's actually a make race in LLVM itself.  I'll post my fix, but it should also be possible to disable PARALLEL_MAKE for that recipe as a work-around.

DunderRoffe, 2017-01-31:
OK, I will that fix then :)

zeenix, 2017-02-01:
@srwalter thanks for providing the fix. Could you make that a pull-request to main repo?

derekstraka, 2017-02-01:
#145 merged it into master

derekstraka, 2017-02-01:
Assuming that #145 fixes this issue.  Re-open if that isn't the case.

closed 2017-02-01 by derekstraka

zeenix, 2017-02-01:
@derekstraka Awesome, thanks. Seems also pushed to krogoth, which is what we use.