meta-rust/meta-rust #162: libstd-rs build fails on morty

SanchayanMaity, 2017-06-28:
## Version(s) of meta-rust
morty Commit Tag: f5387107af7fdb4558d1886d6f68da5ab8403a48

## Version(s) of poky and/or oe-core
oe-core morty

## Expected result
Build of a simple rust hello world project to succeed

## Actual result
Build failure of libstd-rs in the process of building hello world project

## Steps to reproduce
1. Used the below bitbake recipe generated with cargo bitbake to build a simple hello world program for a iMX6 ARM target

```
# Auto-Generated by cargo-bitbake 0.3.6
#
inherit cargo

# If this is git based prefer versioned ones if they exist
# DEFAULT_PREFERENCE = "-1"

# how to get hello_cargo could be as easy as but default to a git checkout:
# SRC_URI += "crate://crates.io/hello_cargo/0.1.0"
SRC_URI += "git://github.com/SanchayanMaity/hello_world_metarust.git;protocol=https"
SRCREV = "04660d0bfce25c2c4c45ec35b47edc56ac0249be"
S = "${WORKDIR}/git"
CARGO_SRC_DIR=""
PV_append = ".AUTOINC+04660d0bfc"

# please note if you have entries that do not begin with crate://
# you must change them to how that package can be fetched
SRC_URI += " \
"

# FIXME: update generateme with the real MD5 of the license file
LIC_FILES_CHKSUM=" \
file://GPL-3.0;md5=generateme \
"

SUMMARY = "Hello world OE meta-rust test"
HOMEPAGE = "sanchayanmaity.github.io"
LICENSE = "CLOSED"

# includes this file if it exists but does not fail
# this is useful for anything you may want to override from
# what cargo-bitbake generates.
include hello_cargo.inc

```
2. Build fails with the following error log

```
ERROR: libstd-rs-1.15.1-r0 do_compile: Function failed: do_compile (log file is located at /build/mas/morty/build/tmp-glibc/work/armv7at2hf-vfp-angstrom-linux-gnueabi/libstd-rs/1.15.1-r0/temp/log.do_compile.31966)
ERROR: Logfile of failure stored in: /build/mas/morty/build/tmp-glibc/work/armv7at2hf-vfp-angstrom-linux-gnueabi/libstd-rs/1.15.1-r0/temp/log.do_compile.31966
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /build/mas/morty/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/cargo
| NOTE: rustc = /build/mas/morty/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/rustc
| NOTE: cargo build -v --target arm-angstrom-linux-gnueabi --release --features 'jemalloc panic-unwind'
| cargo: error while loading shared libraries: libgit2.so.24: cannot open shared object file: No such file or directory
| WARNING: /build/mas/morty/build/tmp-glibc/work/armv7at2hf-vfp-angstrom-linux-gnueabi/libstd-rs/1.15.1-r0/temp/run.do_compile.31966:1 exit 127 from '"cargo" build -v --target arm-angstrom-linux-gnueabi --release --features 'jemalloc panic-unwind' "$@"'
| ERROR: Function failed: do_compile (log file is located at /build/mas/morty/build/tmp-glibc/work/armv7at2hf-vfp-angstrom-linux-gnueabi/libstd-rs/1.15.1-r0/temp/log.do_compile.31966)
ERROR: Task (/build/mas/morty/build/../layers/meta-rust/recipes-devtools/rust/libstd-rs_1.15.1.bb:do_compile) failed with exit code '1'
```
Tried adding a DEPENDS on libgit2 to the libstd-rs recipe but that did not work out. Also looked at Issue #81 but not sure if what I am doing is exactly related to that.

derekstraka, 2017-06-28:
CI build appears to work for qemuarm (ARM9EJS IIRC): https://ci.asterius.io/job/meta-rust-nightly/job/morty/16/

cardoe, 2017-09-29:
So the issue was that libgit2.so.24 was missing from the LD_LIBRARY_PATH of the cargo binary from the cargo-native package. We have dependencies set so I'm not sure why yocto wouldn't have pulled that in.

cardoe, 2018-05-22:
Any further info here or can this be closed?