meta-rust/meta-rust #163: 'libstd' Build Failure on Morty Branch.

robertsonb, 2017-09-07:
Yocto itself is pretty new to me, so I might be doing something wrong here. In any event...

## Version(s) of meta-rust

HEAD on "morty" branch.

## Version(s) of poky and/or oe-core

Poky revision is @ 6a1f33cc40bfac33cf030fe41e1a8efd1e5fad6f.

Poky: v2.2.1
Morty: 16.0.1

## Expected result

Expectation is that libstd builds and links correctly.

## Actual result

Builds failed with the following output:

|   = note: /workdir/voiceswitch-vs59a/build/tmp/work/armv7at2hf-neon-fslc-linux-gnueabi/libstd-rs/1.15.1-r0/rustc-1.15.1-src/arm-fslc-linux-gnueabi/release/deps/std.0.o: In function `_$LT$alloc..arc..Arc$LT$T$GT$$GT$::drop_slow::h033fec3c07bd5dcc':
| std.cgu-0.rs:(.text._ZN33_$LT$alloc..arc..Arc$LT$T$GT$$GT$9drop_slow17h033fec3c07bd5dccE+0x58): relocation truncated to fit: R_ARM_THM_JUMP11 against symbol `__rust_deallocate' defined in .text.__rust_deallocate section in /tmp/rustc.IJAes26F2x4F/liballoc_system-2f80c92c5b4c2811.rlib(alloc_system-2f80c92c5b4c2811.0.o)

For more output see:
[meta_rust_morty_libstd_link_failure.txt](https://github.com/meta-rust/meta-rust/files/1286038/meta_rust_morty_libstd_link_failure.txt)

My target is an iMX6 (which looks like it has T2 support). 

Looking into it more it seems like the JUMP is too far, so the linker fails when it goes to fill in the addresses...?

## Steps to reproduce

Add meta-rust layer and then just try and build an image.

rust-llvm-native will build successfully, cargo builds as well, then libstd fails.

cardoe, 2017-09-10:
`R_ARM_THM_JUMP11` does that imply you are using Thumb? Its possible that the jump offset allowed with Thumb is smaller. I'm not terribly familiar with Thumb. @derekstraka any ideas?

robertsonb, 2017-09-10:
Yes, its building with Thumb.

I have been working on this more this morning and come up with a few things.

Had issue open over at the Repo as well which Im keeping up to date: https://github.com/rust-lang/rust/issues/44458

robertsonb, 2017-09-10:
If I remove the "--release" flag from `cargo.bbclass` so no optimizations are performed, I get only 2 errors... the same error but originating from the `libcore-*.rlib` artifact.

If dump the assembly from that rlib and look at unresolved `b.n` instructions the only ones I see are the 2 remaining errors.

cardoe, 2017-09-10:
There is a Rust 1.18 branch that you can try. It's in a PR or under my github repo. I'm working on updating the Rust version which might pull in a new LLVM that will help.

derekstraka, 2017-09-10:
You might try disabling thumb `ARM_INSTRUCTION_SET = "arm"` as a work around?

robertsonb, 2017-09-10:
@cardoe I actually have done that. I took your 1.18 branch and pulled that into Morty. So Im building w/ 1.18 now.

@derekstraka Thanks. I didn't know about the Yocto option. I'll look into using that build recipes/layers.

robertsonb, 2017-09-10:
Taking a cue from @derekstraka, Im looking for ways to disable Thumb entirely, but without rebuilding the meta-rust layer.

Im looking at the `rust.inc` file right now. Can I change something in there that would prevent `rustc` from emitting thumb code (-mnothumb)? I would much rather do that than set Yocto config values.

derekstraka, 2017-09-10:
You should be able to just add that to the individual recipe that is
causing problems.

robertsonb, 2017-09-10:
Yeah, I was looking at `libstd_1.18.0.bb` and it has CARGO_BUILD_FLAGS. The problem is getting Cargo to forward params the `rustc`. 

Looking at setting RUSTFLAGS in cargo.bbclass, as it looks like thats what actually invokes Cargo in the layer.

robertsonb, 2017-09-11:
I was able to workaround this by removing the -mthumb option completely and rebuilding the compiler.

Right now Im not running Cargo with the --release flag, just doing Debug builds.

closed 2017-09-11 by robertsonb