meta-rust/meta-rust #172: cargo built binary rpath incorrect on morty

willnewton, 2017-10-05:
## Version(s) of meta-rust

morty f538710

## Version(s) of poky and/or oe-core

morty

## Expected result

I build a binary with cargo.bbclass and RUSTFLAGS += "-C prefer-dynamic" and get a dynamic linked executable as expected.

## Actual result

When I run the executable I get a failure to find libstd.so. This is because DT_RUNPATH does not include /usr/lib/rust. I noticed that rust-bin.bbclass includes a function do_rust_bin_fixups that will fix this for me, and copying this function into my recipe works.

It feels like there should be some way to enable this functionality by default but I am not sure how to go about it.

cardoe, 2017-10-10:
I'm certainly game for improvements. I'm not sure the "Yocto Way" of doing this. @derekstraka any ideas?

tylerwhall, 2017-10-10:
I made a local change to make the rpath handling in rust-bin a bit better, but hadn't upstreamed since rust-bin isn't really the desired/supported/tested workflow.

https://github.com/tylerwhall/meta-rust/commit/9a0b3b65a5880d59f62e51902368776c3d3086b0

The patch is less of a hack than it is currently, by setting the rpath on the link-line rather than using patchelf. Still not ideal though, because it's passing flags to ld directly and there's the inconvenience that only one instance of `-C link-args=` is accepted by rustc.

Cargo doesn't seem to have rpath logic (other than a bool to pass `-C rpath`), so whatever solution should apply to both workflows. Probably we'd have to get rustc to do the right thing or accept passing `link-args`. Currently I believe rustc uses a compile-time constant (that's incorrect) but I could be wrong.

cardoe, 2017-10-31:
Let's include what we can upstream.

Fulkerson, 2017-12-02:
Another possibility is to not use rpath but instead use `/etc/ld.so.conf`. We could do what `meta/recipes-extended/bash/bash.inc` does to append to `/etc/shells`

I know using rpath is usually frowned upon for various reasons.

cardoe, 2018-05-22:
@tylerwhall Did you upstream the fix in a newer branch? @Fulkerson is this resolved through a newer branch or some of your other work?

Fulkerson, 2018-05-22:
@cardoe No I did some testing but couldn't get it to work so I abandoned it.