meta-rust/meta-rust #173: The file /usr/include/llvm/LinkAllIR.h is installed by both rust-llvm-native and clang-native, aborting

zzeroo, 2017-10-08:
## Version(s) of meta-rust
7f80e6d68b48a0a2446a606dd35901db4505d61a (HEAD -> master, upstream/master)

## Version(s) of poky and/or oe-core
```
Build Configuration:
BB_VERSION        = "1.35.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "bananapro"
DISTRO            = "eisenhardt"
DISTRO_VERSION    = "1.0"
TUNE_FEATURES     = "arm armv7ve vfp neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta
meta-poky
meta-yocto-bsp    = "master:fcdfe09d9cc4e2977bd602dfd18f729cd9f6d73d"
meta-clang        = "master:5176136fe731f0e8e526fe7252e607dbafb3ac92"
meta-bsp_lemaker  = "development:dad114e049e7137691aa75d4c7fee77619fe75ee"
meta-xmz-mod-touch = "development:678252ce7693b4c64cdb10979e25d1e385c22e64"
meta-oe           = "master:5c9a4a08844c215ebd6f2146f50de753b8e7ecef"
meta-rust         = "master:7f80e6d68b48a0a2446a606dd35901db4505d61a"
```

## Expected result
No error

## Actual result
Error `The file /usr/include/llvm/LinkAllIR.h is installed by both rust-llvm-native and clang-native, aborting`

## Steps to reproduce
Include the `meta-rust` and `meta-clang` in your local bblayers. Then create a recipe with `DEPENDS += "clang-native"` and `bitbake` this.

I'm unsure where to report this issue. So I will here and in meta-clang

cardoe, 2017-10-10:
We can probably get away with no-oping the install of that package for header files since we don't really want people using the Rust version of LLVM. The only reason we packaged this up is that there wasn't an LLVM that was up to date enough being shipped with Yocto.

derekstraka, 2017-10-10:
At some point we should try to grab the yocto LLVM and use that, but for now we can just do an install append to pick off the offending files.

zzeroo, 2017-10-11:
My solution so far isn't working ...

```bash
# rust-llvm_%.bbappend
do_install_append() {
  rm -rf  ${D}${includedir}/llvm/LinkAllIR.h
  rm -rf  ${D}${includedir}/llvm/PassInfo.h
}
```
results in this error.
```bash
cargo:warning=/home/build/shared/build-bananapro/tmp/work/x86_64-linux/rust-native/1.20.0-r0/recipe-sysroot-native/usr/include/llvm/PassSupport.h:26:27: fatal error: llvm/PassInfo.h: No such file or directory
```
Without rm PassInfo.h I tigger the original error from above.

jmesmon, 2017-10-17:
Yep, rust needs that file to build :) .

Realistically, we need to either:
 - install rust-llvm so it doesn't collide with llvm (should be the same manner multiple llvm versions are supported in some distros)
 - use plain llvm for rust.

cardoe, 2017-10-18:
We probably should do `PROVIDES = "llvm"` in our rust-llvm. The fact that meta-clang is building LLVM inside their clang recipe is weird to me. I always though clang was another package that built using an LLVM.

Fulkerson, 2017-10-21:
I don't think we want to use plain llvm. The rust fork contain some patches: https://github.com/rust-lang/rust/issues/42389.

We'd be better of simply placing it somewhere else.

cardoe, 2017-10-30:
Agreed. I'm going to close #175.

zzeroo, 2017-10-31:
I want to try to refocus this issue. Because I think we run into the [XY Pproblem](https://meta.stackexchange.com/questions/66377/what-is-the-xy-problem).

A [project of mine](https://github.com/zzeroo/libmodbus-rs) uses Rust and bindgen. bindgen itself depends on libclang-3.9++

What would you suggest? What should I write in the recipe to make the libclang available to my [build.rs](https://github.com/zzeroo/libmodbus-rs/blob/master/libmodbus-sys/build.rs)?

jmesmon, 2017-11-30:
I don't think this is an XY problem at all. We really need to just build rust's llvm so it installs to a different name to avoid collisions between the 2 variants.

I'm reopening as this needs fixing one way or another in meta-rust.

cosmo0920, 2018-04-02:
Any update on this?
I've also hit this issue during working on building our customized [meta-browser](https://github.com/webdino/meta-browser/tree/firefox-60-wip).
In firefox ESR60, it depends on Rust cross compiler and libclang.so for generating cross Rust crates.

derekstraka, 2018-04-02:
@cosmo0920 - AFAIK no one is actively working on this ticket.

PR #199 (2018-07-07): Various fixes, Clang, TCLIBC, DEBUG_BUILD, rocko 
This is a few smaller fixes and tweaks I've come up with.

First is if you set DEBUG_BUILD="1" in a cargo recipe, cargo will build without --release. This mimics DEBUG_BUILD pretty well (it sets "-O0 -g" etc in normal builds). set DEBUG_BUILD_pn-whateverpackage = "1" in conf/local.conf for much shorter compile times.

Second is a change which installs rust-llvm into a subdir instead of ${prefix} as usual. It makes it possible to use meta-rust and meta-clang together. It should also fix #173.

Third adds the possibility to use TCLIBC="musl". I.e [... cut]
closed 2018-07-07 by derekstraka