meta-rust/meta-rust #183: [request] Backport new rust to morty

peacememories, 2017-12-01:
Right now the morty branch is running rust 1.14
It would be nice to have it running 1.21 since we are not able to switch to pyro with our project at the moment.

Thanks for providing this layer :)

cardoe, 2017-12-01:
You should have 1.15.1 in the morty branch currently. We have a supported matrix that can be seen at: https://github.com/meta-rust/meta-rust/wiki I'm afraid the backport isn't really that trivial and will require a bit of work which I'm not sure if one of the existing maintainers will want to do. As such I don't think there's a reason why we wouldn't accept a patch if your organization wanted to put the work in since its directly benefiting your organization's business needs.

peacememories, 2018-05-23:
Our team has since been able to move to rocko (and I've moved on to a different job), so you don't need to keep this open on my account. Sorry for the trouble :)

cardoe, 2018-05-23:
Thanks for the update!

closed 2018-05-23 by cardoe