meta-rust/meta-rust #186: Create rocko branch

hmvp, 2018-02-01:
## Version(s) of meta-rust
rocko
## Version(s) of poky and/or oe-core
rocko
## Expected result
I expect to be able to build rocko based images using the rocko branch for all meta layers
## Actual result
I have to make an exception and use the master branch?
## Steps to reproduce
-

hmvp, 2018-02-01:
One actual issue with using master for a Rocko release is that a lot of getVar calls are missing their expand argument. It seems they were remove before pyro because they were thought to default to True. However in Rocko it is a hard error to miss that argument and in Pyro the default is still False...

derekstraka, 2018-02-02:
I created a preliminary rocko branch based on what is in master.

hmvp, 2018-02-05:
Everything seems to work so far.

derekstraka, 2018-02-05:
Completed.

closed 2018-02-05 by derekstraka