meta-rust/meta-rust #195: devtool modify does not fetch crate://

Fulkerson, 2018-05-19:
## Version(s) of meta-rust
All

## Version(s) of poky and/or oe-core
All

## Expected result
devtool is a tool which fetches and redirects the source of a package to an external directory in *workspace/sources/package*. Modifications to the source can be made and compilation is performed using `devtool build package`. It is more or less a necessity when developing code for an embedded system.

## Actual result

The source to the package is fetched properly. However the crates specified in the recipe are not properly fetched, i.e. *cargo_home/bitbake* is not created resulting in error messages when trying to build.

```
$ cat /path/to/build/tmp/work/custom-poky-linux/rustfmt/0.8.0-r0/temp/log.do_compile.12811

DEBUG: Executing python function externalsrc_compile_prefunc
NOTE: rustfmt: compiling from external source tree /path/to/build/workspace/sources/rustfmt
DEBUG: Python function externalsrc_compile_prefunc finished
DEBUG: Executing shell function do_compile
NOTE: cargo = /path/to/build/tmp/work/custom-poky-linux/rustfmt/0.8.0-r0/recipe-sysroot-native/usr/bin/cargo
NOTE: rustc = 
NOTE: cargo build -v --target x86_64-poky-linux --release 
error: failed to load source for a dependency on `diff`

Caused by:
  Unable to update registry `https://github.com/rust-lang/crates.io-index`

Caused by:
  failed to update replaced source registry `https://github.com/rust-lang/crates.io-index`

Caused by:
  failed to read root of directory source: /path/to/build/tmp/work/custom-poky-linux/rustfmt/0.8.0-r0/cargo_home/bitbake

Caused by:
  No such file or directory (os error 2)
WARNING: exit code 101 from a shell command.
ERROR: Function failed: do_compile (log file is located at /path/to/build/tmp/work/custom-poky-linux/rustfmt/0.8.0-r0/temp/log.do_compile.12811)
```

## Steps to reproduce

```
$ devtool modify rustfmt
$ devtool build rustfmt
```

siredmar, 2019-03-07:
I have an build issue for balena-raspberry based on meta-rust sumo branch. 
I have discussed with one balena guy and shared my findings so far:
https://github.com/balena-os/balena-raspberrypi/issues/261

@srwalter  @derekstraka 
Can any of you give me an idea what could go wrong here?

srwalter, 2019-03-07:
@siredmar a couple thoughts.  First, just to clarify, are you using devtool?  This issue was specifically around devtool, but I didn't see any mention of that in the other issue.

Ubuntu 18.04 vs. 16.04 shouldn't make a difference.  I'm using 18.04 with sumo and I can build cargo stuff fine, with devtool and without.  The only non-standard thing about my configuration that I'm aware of is I always setup /bin/sh so that it uses bash, rather than dash.  "sudo dpkg-reconfigure dash" and select "no" if you want to try that.

In one of your comments, you said "CARGO_HOME is balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/cargo_home/".  That seems wrong.  cargo_common.bbclass sets CARGO_HOME to ${WORKDIR}/cargo_home; the value you gave indicates that WORKDIR has "git" in it, which I wouldn't expect.

Do you have any local modifications at all?  Adding a crate to Cargo.toml without updating the recipe could result in similar fetch failures.

Another test you could try is building just upstream poky with meta-rust and see if "bitbake cargo-native" gives you trouble.  If not that would point to something being weird in the balena-raspberrypi layer.

siredmar, 2019-03-07:
@srwalter no, it is a clean vanilla balena build on Ubuntu 18.04. 
I am not using anything like devtool - at least not that i know of. 
As far as is can estimate, balena wraps the bitbake command around a own command line tool called barys.

What i also did was successfully building the balena-raspberrypi repository with the current master of meta-rust on Ubuntu 18.04 - however i did not test if the image acutally boots up. Just wanted to test if the cargo-native build problem was gone.

Just for your information: I put up a vanilla Ubuntu 16.04.06 and did a clean build of balena-raspberrypi with no isues at all.

siredmar, 2019-03-08:
> Another test you could try is building just upstream poky with meta-rust and see if "bitbake cargo-native" gives you trouble. If not that would point to something being weird in the balena-raspberrypi layer.

I am doing a test right now with the poky from balena-raspberry and the unmodified sumo meta-rust layer used by balena-raspberrypi (and of course meta-openembedded/meta-oe as a dependency to meta-rust) 

So far: 
I am running 
```
$ cd balena-raspberrypi/layers/poky
$ MACHINE=qemux86 bitbake cargo-native 
```

```
$ pwd
/opt/projects/marblerun/marblerun-balena/build/balena-raspberrypi/layers/poky/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0

$ ls git                                                                     
appveyor.yml  ARCHITECTURE.md  Cargo.toml  CONTRIBUTING.md  LICENSE-APACHE  LICENSE-MIT  LICENSE-THIRD-PARTY  README.md  rustfmt.toml  src  tests

$ ls cargo_home/bitbake                                               
advapi32-sys-0.2.0         flate2-1.0.1                matches-0.1.6           regex-0.1.80                   termion-1.5.1
aho-corasick-0.5.3         fnv-1.0.6                   memchr-0.1.11           regex-0.2.6                    thread-id-2.0.0
aho-corasick-0.6.4         foreign-types-0.3.2         memchr-1.0.2            regex-syntax-0.3.9             thread_local-0.2.7
atty-0.2.6                 foreign-types-shared-0.1.1  memchr-2.0.1            regex-syntax-0.4.2             thread_local-0.3.5
backtrace-0.3.5            fs2-0.4.3                   miniz-sys-0.1.10        remove_dir_all-0.3.0           toml-0.4.5
backtrace-sys-0.1.16       fuchsia-zircon-0.3.3        miow-0.2.1              rustc-demangle-0.1.6           unicode-bidi-0.3.4
bitflags-0.9.1             fuchsia-zircon-sys-0.3.3    net2-0.2.31             rustc-serialize-0.3.24         unicode-normalization-0.1.5
bitflags-1.0.1             git2-0.6.11                 num-0.1.42              same-file-0.1.3                unicode-xid-0.0.4
bufstream-0.1.3            git2-curl-0.7.0             num-bigint-0.1.43       schannel-0.1.10                unreachable-1.0.0
cc-1.0.4                   glob-0.2.11                 num-complex-0.1.42      scoped-tls-0.1.0               url-1.6.0
cfg-if-0.1.2               globset-0.2.1               num_cpus-1.8.0          scopeguard-0.1.2               userenv-sys-0.2.0
cmake-0.1.29               hamcrest-0.1.1              num-integer-0.1.36      semver-0.8.0                   utf8-ranges-0.1.3
commoncrypto-0.2.0         hex-0.2.0                   num-iter-0.1.35         semver-parser-0.7.0            utf8-ranges-1.0.0
commoncrypto-sys-0.2.0     home-0.3.0                  num-rational-0.1.42     serde-1.0.27                   vcpkg-0.2.2
core-foundation-0.4.6      idna-0.1.4                  num-traits-0.1.43       serde_derive-1.0.27            void-1.0.2
core-foundation-sys-0.4.6  ignore-0.2.2                num-traits-0.2.0        serde_derive_internals-0.19.0  walkdir-1.0.7
crossbeam-0.2.12           itoa-0.3.4                  openssl-0.9.24          serde_ignored-0.0.4            winapi-0.2.8
crossbeam-0.3.2            jobserver-0.1.9             openssl-probe-0.1.2     serde_json-1.0.9               winapi-0.3.4
crypto-hash-0.3.0          kernel32-sys-0.2.2          openssl-sys-0.9.26      shell-escape-0.1.3             winapi-build-0.1.1
curl-0.4.11                lazy_static-0.2.11          percent-encoding-1.0.1  socket2-0.3.1                  winapi-i686-pc-windows-gnu-0.4.0
curl-sys-0.4.1             lazy_static-1.0.0           pkg-config-0.3.9        strsim-0.6.0                   winapi-x86_64-pc-windows-gnu-0.4.0
docopt-0.8.3               libc-0.2.36                 psapi-sys-0.1.1         syn-0.11.11                    wincolor-0.1.6
dtoa-0.4.2                 libgit2-sys-0.6.19          quote-0.3.15            synom-0.11.3                   ws2_32-sys-0.2.1
env_logger-0.4.3           libssh2-sys-0.2.6           rand-0.3.22             synstructure-0.6.1
failure-0.1.1              libz-sys-1.0.18             rand-0.4.2              tar-0.4.14
failure_derive-0.1.1       log-0.3.9                   redox_syscall-0.1.37    tempdir-0.3.6
filetime-0.1.15            log-0.4.1                   redox_termios-0.1.1     termcolor-0.3.4
```
There is no cargo_home/config file present
The build isn't finished yet, but it seems the crates have been successfully downloaded via bitbake