meta-rust/meta-rust #209: cargo-native build failed

mark2b, 2018-10-07:
## Version(s) of meta-rust
sumo
## Version(s) of poky and/or oe-core
2.5 sumo
## Expected result
cargo-native compiled successfully 
## Actual result
cargo-native build failed because of transitive dependency of newest version zlib-sys-1.0.23
**cannot find function `read_to_string` in module `fs`**
**read_to_string** exists starting from rust 1.26, but sumo installs rust earlier versions
## Steps to reproduce
bitbake cargo-native

```
ERROR: cargo-native-0.25.0-r0 do_compile: Function failed: do_compile (log file is located at /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.895)
ERROR: Logfile of failure stored in: /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.895
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo
| NOTE: rustc =
| NOTE: /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo build -v --target x86_64-linux --release
|     Updating registry `https://github.com/rust-lang/crates.io-index`
|  Downloading serde v1.0.79
|  Downloading filetime v0.1.15
|  Downloading tempdir v0.3.7
|  Downloading same-file v0.1.3
|  Downloading tar v0.4.17
|  Downloading serde_derive v1.0.79
|  Downloading failure v0.1.2
|  Downloading fs2 v0.4.3
|  Downloading toml v0.4.8
|  Downloading crossbeam v0.3.2
|  Downloading git2 v0.6.11
|  Downloading num_cpus v1.8.0
|  Downloading termcolor v0.3.6
|  Downloading home v0.3.3
|  Downloading jobserver v0.1.11
|  Downloading git2-curl v0.7.0
|  Downloading semver v0.8.0
|  Downloading flate2 v1.0.3
|  Downloading libgit2-sys v0.6.19
|  Downloading crypto-hash v0.3.1
|  Downloading atty v0.2.11
|  Downloading curl v0.4.18
|  Downloading libc v0.2.43
|  Downloading glob v0.2.11
|  Downloading serde_ignored v0.0.4
|  Downloading hex v0.2.0
|  Downloading url v1.7.1
|  Downloading log v0.3.9
|  Downloading shell-escape v0.1.4
|  Downloading ignore v0.2.2
|  Downloading scoped-tls v0.1.2
|  Downloading serde_json v1.0.32
|  Downloading env_logger v0.4.3
|  Downloading docopt v0.8.3
|  Downloading cfg-if v0.1.5
|  Downloading rand v0.4.3
|  Downloading remove_dir_all v0.5.1
|  Downloading filetime v0.2.1
|  Downloading quote v0.6.8
|  Downloading proc-macro2 v0.4.20
|  Downloading syn v0.15.8
|  Downloading unicode-xid v0.1.0
|  Downloading backtrace v0.3.9
|  Downloading failure_derive v0.1.2
|  Downloading rustc-demangle v0.1.9
|  Downloading syn v0.14.9
|  Downloading synstructure v0.9.0
|  Downloading bitflags v0.9.1
|  Downloading percent-encoding v1.0.1
|  Downloading matches v0.1.8
|  Downloading idna v0.1.5
|  Downloading unicode-bidi v0.3.4
|  Downloading unicode-normalization v0.1.7
|  Downloading libz-sys v1.0.23 - latest version uses API of lates rust version
|  Downloading libssh2-sys v0.2.11
|  Downloading curl-sys v0.4.12
|  Downloading pkg-config v0.3.14
|  Downloading cc v1.0.25
|  Downloading cmake v0.1.35
|  Downloading itoa v0.4.3
|  Downloading ryu v0.2.6
|  Downloading socket2 v0.3.8
|  Downloading log v0.4.5
|  Downloading semver-parser v0.7.0
|  Downloading miniz-sys v0.1.10
|  Downloading hex v0.3.2
|  Downloading thread_local v0.3.6
|  Downloading memchr v1.0.2
|  Downloading lazy_static v0.2.11
|  Downloading walkdir v1.0.7
|  Downloading crossbeam v0.2.12
|  Downloading globset v0.2.1
|  Downloading regex v0.2.11
|  Downloading lazy_static v1.1.0
|  Downloading version_check v0.1.5
|  Downloading memchr v2.1.0
|  Downloading aho-corasick v0.6.8
|  Downloading fnv v1.0.6
|  Downloading regex-syntax v0.5.6
|  Downloading utf8-ranges v1.0.1
|  Downloading ucd-util v0.1.1
|  Downloading strsim v0.6.0
|  Downloading backtrace-sys v0.1.24
|  Downloading openssl-probe v0.1.2
|  Downloading openssl-sys v0.9.36
|  Downloading openssl v0.10.12
|  Downloading bitflags v1.0.4
|  Downloading foreign-types v0.3.2
|  Downloading foreign-types-shared v0.1.1
|    Compiling hex v0.3.2
|    Compiling ucd-util v0.1.1
|    Compiling failure_derive v0.1.2
|    Compiling lazy_static v0.2.11
| 

...................

|    Compiling libz-sys v1.0.23
|      Running `rustc --crate-name build_script_build /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libz-sys-1.0.23/build.rs --crate-type bin --emit=dep-info,link -C opt-level=3 -C metadata=ec7a812521c0ed10 -C extra-filename=-ec7a812521c0ed10 --out-dir /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/release/build/libz-sys-ec7a812521c0ed10 -C linker=/workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-ccld -L dependency=/workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/release/deps --extern pkg_config=/workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/release/deps/libpkg_config-3d4e1cdfd28e97cf.rlib --extern cc=/workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/release/deps/libcc-f2d1c26f21387ba1.rlib --cap-lints allow`
| error[E0425]: cannot find function `read_to_string` in module `fs`
|    --> /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libz-sys-1.0.23/build.rs:118:13
|     |
| 118 |         fs::read_to_string("src/zlib/zlib.pc.in")
|     |             ^^^^^^^^^^^^^^ did you mean `read_string`?
| 
| error: use of unstable library feature 'fs_read_write' (see issue #46588)
|    --> /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libz-sys-1.0.23/build.rs:116:5
|     |
| 116 |     fs::write(
|     |     ^^^^^^^^^
| 
| error[E0425]: cannot find function `read_to_string` in module `fs`
|   --> /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:99:26
|    |
| 99 |         let config = fs::read_to_string("libssh2/src/libssh2_config_cmake.h.in")
|    |                          ^^^^^^^^^^^^^^ did you mean `read_string`?
| 
| error[E0425]: cannot find function `read_to_string` in module `fs`
|    --> /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:124:24
|     |
| 124 |     let libssh2h = fs::read_to_string("libssh2/include/libssh2.h").unwrap();
|     |                        ^^^^^^^^^^^^^^ did you mean `read_string`?
| 
| error[E0425]: cannot find function `read_to_string` in module `fs`
|    --> /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:134:13
|     |
| 134 |         fs::read_to_string("libssh2/libssh2.pc.in")
|     |             ^^^^^^^^^^^^^^ did you mean `read_string`?
| 
| error: use of unstable library feature 'fs_read_write' (see issue #46588)
|    --> /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:105:9
|     |
| 105 |         fs::write(build.join("libssh2_config.h"), &config).unwrap();
|     |         ^^^^^^^^^
| 
| error: use of unstable library feature 'fs_read_write' (see issue #46588)
|    --> /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:132:5
|     |
| 132 |     fs::write(
|     |     ^^^^^^^^^
| 
| error: aborting due to 2 previous errors

```

jmesmon, 2018-10-14:
This is interesting. From your description, it seems like some crate may have not had a strong enough version requirement, and/or `Cargo.lock` is not being used/respected, and/or some crate introduced a breaking change (higher rustc requirement) when it shouldn't have.

I'd be curious why the `Cargo.lock` (which should have prevented this from happening) isn't being respected. `cargo` does include a `Cargo.lock`, right? Maybe we're somehow allowing it to be updated?

Another realistic "quick fix" may just be upgrading the `rustc` we ship.

mark2b, 2018-10-14:
Actually I don't see Cargo.lock in the repository. 

Concerning upgrading the `rustc`
I've already tried master branch and copy latest rust recipes from **master** to **sumo**.
But there is another problem 
Build failed by linking with **libssh2** / **libssh2_sys**

[output.txt](https://github.com/meta-rust/meta-rust/files/2476081/output.txt)
`
|   = note: /workspace/tmp/hosttools/ld: /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/x86_64-linux/release/deps/cargo-f657a3b6e33a1544: hidden symbol `libssh2_init' in /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/x86_64-linux/release/deps/liblibssh2_sys-178179f5ac1417f8.rlib(global.o) is referenced by DSO
|           /workspace/tmp/hosttools/ld: final link failed: Bad value
|           collect2: error: ld returned 1 exit status
| 
`
Below is partial log, and full log is attached. 
`
Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "white-rpi3"
DISTRO               = "white-central"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "sumo:36d5cee56bd389ecbe66dbfad9e5d57c13b56b95"
meta-raspberrypi     = "sumo:2d40b000021bc8a9ef7f329ed0ad410f8d227b97"
meta-mender-core     
meta-mender-raspberrypi = "sumo:ab9c2cb52b522f7734909ae9974e2ca18824fc82"
meta-oe              
meta-networking      = "sumo:2bb21ef27c4b0c9d52d30b3b2c5a0160fd02b966"
meta-rust            = "sumo:563cf902d582918602255fee567b302101bbf6e5"
meta-python          = "sumo:2bb21ef27c4b0c9d52d30b3b2c5a0160fd02b966"
meta-white           = "<unknown>:<unknown>"

Initialising tasks: 100% |##############################################################################################################################| Time: 0:00:02
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: cargo-native-0.25.0-r0 do_compile: Function failed: do_compile (log file is located at /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.154)
ERROR: Logfile of failure stored in: /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.154
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo
| NOTE: rustc =
| NOTE: /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo build -v --target x86_64-linux --release
|        Fresh cfg-if v0.1.5
|        Fresh utf8-ranges v1.0.1
|        Fresh cc v1.0.25
|        Fresh version_check v0.1.5
|        Fresh shell-escape v0.1.4
|        Fresh matches v0.1.8
|        Fresh termcolor v0.3.6
|        Fresh crossbeam v0.3.2
|        Fresh strsim v0.6.0
|        Fresh crossbeam v0.2.12
|        Fresh foreign-types-shared v0.1.1
|        Fresh hex v0.2.0
|        Fresh libc v0.2.43
|        Fresh glob v0.2.11
|        Fresh rustc-demangle v0.1.9
|        Fresh ucd-util v0.1.1
|        Fresh bitflags v1.0.4
|        Fresh fnv v1.0.6
|        Fresh itoa v0.4.3
|        Fresh percent-encoding v1.0.1
|        Fresh bitflags v0.9.1
|        Fresh same-file v0.1.3
|        Fresh lazy_static v0.2.11
[... 61 lines cut ...]
|        Fresh env_logger v0.4.3
|        Fresh git2 v0.6.11
|        Fresh failure v0.1.2
|        Fresh ignore v0.2.2
|        Fresh git2-curl v0.7.0
|        Fresh crates-io v0.14.0 (file:///workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/src/crates-io)
|    Compiling cargo v0.25.0 (file:///workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git)
|      Running `rustc --crate-name cargo src/bin/cargo.rs --crate-type bin --emit=dep-info,link -C opt-level=3 -C metadata=f657a3b6e33a1544 -C extra-filename=-f657a3b6

TOO MANY OUTPUT
../../../usr/lib/pkgconfig/../../../usr/lib -L native=/workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/x86_64-linux/release/build/libssh2-sys-ca059eed68c54e49/out/build`
| error: linking with `/workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-ccld` failed: exit code: 1
TOO MANY OUTPUT
" "util" "-l" "util" "-l" "dl" "-l" "rt" "-l" "pthread" "-l" "gcc_s" "-l" "c" "-l" "m" "-l" "rt" "-l" "pthread" "-l" "util" "-l" "util"
|   = note: /workspace/tmp/hosttools/ld: /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/x86_64-linux/release/deps/cargo-f657a3b6e33a1544: hidden symbol `libssh2_init' in /workspace/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/x86_64-linux/release/deps/liblibssh2_sys-178179f5ac1417f8.rlib(global.o) is referenced by DSO
|           /workspace/tmp/hosttools/ld: final link failed: Bad value
|           collect2: error: ld returned 1 exit status
| 
| 
| error: aborting due to previous error
| 
| error: Could not compile `cargo`.
| 

`

jcreekmore, 2018-10-15:
Here is my understanding of the issue: rust-native and cargo-native both built successfully, but when you are trying to use the cargo-native to build some other Rust-written utility, it fails to build because there is either not a Cargo.lock file or it is dependent on crates that are dependent on newer versions of Rust (than the 1.24.x that is included in sumo). 

So, it isn't an issue with cargo-native and the sumo meta-rust layer other than it has an older version of Rust than you require.

Concerning:
> Build failed by linking with libssh2 / libssh2_sys

I have run into something similar to that in the past, but I don't know enough BitBake-foo to figure out how to force those libraries to be available for build-time, since (at least in the case I had) they were required by a build-script and I couldn't figure out how to force them to be included for the execution environment for the build.rs binary. Maybe @derekstraka might know?

mark2b, 2018-10-15:
Problem with build of cargo-native. Just run `bitbake cargo-native` or `bitbake rust` and it will failed. 
It worked for fine me before two / three months until I wiped out workspace for clean build. 
There are many base crates (cargo dependencies) uses the latest rust API in build.rs. All these changes are two/three months old.

jcreekmore, 2018-10-15:
I literally just built it today by running (in the sumo branch): `./scripts/fetch.sh sumo` and `./scripts/build.sh` after removing my build directory. Those scripts go through and build up the toolchain and finally build up rustfmt using the built rustc and cargo.

mark2b, 2018-10-17:
I just checked, it really works. you gave me hope
Will check what is wrong in my environment.

mark2b, 2018-10-21:
All my image built successfully in case my Dockerfile inherited from **cardoe/yocto:pyro**
Previously Dockerfile was inherited from **debian:latest** with Yocto prerequisite dependencies. 
This is only the difference I found to make my project build.

otavio, 2018-10-30:
It has indeed some relation with the host system. I tried this with `thud` in two different hosts.

My machine, which is an Arch Linux, it works fine. Our CI server which runs on the cloud using an Ubuntu 16.04 it fails.

```
ossystems-gamepad-image@gamepad-rpi3 |   = note: .../build/tmp/hosttools/ld: .../build/tmp/work/x86_64-linux/cargo-native/0.28.0-r0/git/target/x86_64-linux/release/deps/cargo-7242ceb102b4b088: hidden symbol `libssh2_init' in .../build/tmp/work/x86_64-linux/cargo-native/0.28.0-r0/git/target/x86_64-linux/release/deps/liblibssh2_sys-0f6397a7f07d25ad.rlib(global.o) is referenced by DSO
```

iohe, 2018-11-04:
Hello, i also see the  hidden symbollibssh2_init issue, but my context is a little different.
Using yocto thud,Ubuntu 16.04.3, and on top of meta-rust updated recipes for cargo-0.30 and rust-1.30.0.
For rust-native, it compiles, but i had to change in rust sources tools.rs:353 RustInstaller mode from ToolBootstrap to ToolStd in order for rust-native:do_install to actually finish.

So, i used nm to look into liblibssh2-sys-* and in global.o section, i see a T before libssh2_init , in global.o section. 

Any idea ?

otavio, 2018-11-05:
> Using yocto thud,Ubuntu 16.04.3, and on top of meta-rust updated recipes for cargo-0.30 and rust-1.30.0.

Please send a PR for it, so we all avoid duplicated work.

> So, i used nm to look into liblibssh2-sys-* and in global.o section, i see a T before libssh2_init , in global.o section. It is U (undefined) in libssh2_sys-015deefc16288fe0.libssh2_sys.dwt87r28-cgu.0.rcgu.o section.

Does https://github.com/OSSystems/meta-rust/commit/52814ce1cedc36b66dfbd1fe31be700fa488df07 helps?

otavio, 2018-11-05:
Just to inform that PR #212 fixes the build error for me on my CI. @iohe please rebase your updated recipes for Rust 1.30.0 so we don't duplicate work. I am willing to test it if you want.

JoelColledge, 2018-11-05:
I observed the same errors (on `sumo` and on `6b06061`) using an Ubuntu 18.04 host and #212 fixes the build for me too :+1: 

In case it is of use to anyone - I also found that `cargo-native` version `0.22.0` as specified in branch `sumo` built fine too. Just override with `PREFERRED_VERSION_cargo-native = "0.22.0"`.

PR #212 (2018-11-13): Use libssh2 from system and upgrade cargo to 0.31.0
This pull request enables the use of pkgconfig of libssh2 so it links against Yocto Project library and bumps the cargo utility to 0.31.0 release.

Fixes: #209.
closed 2018-11-13 by derekstraka