meta-rust/meta-rust #218: thud branch

floion, 2018-11-26:
Hi,
is thud support close to be released? I just checked and did not see a `thud` branch created.

Thank you

floion, 2018-11-26:
cc @derekstraka

derekstraka, 2018-11-27:
I'm going to be honest, and say that I have not been actively testing with thud pre-releases.  If you or someone you know has verified meta-rust/master works, I'll make the branch

siredmar, 2019-03-05:
As thud has been released, will there be any effort for a thud branch?

srwalter, 2019-04-08:
FWIW, I'm using the master branch with sumo, so I would expect that the master branch works with thud as well

floion, 2019-09-30:
Hi, so can the `thud` and `warrior` branches be created?

wdehoog, 2019-10-21:
Just to let you know. I am currently using master to build my layer with the thud branches of poky and meta-openembedded.

floion, 2019-10-22:
Thanks, so we'll just use master until these branches are officially created

floion, 2020-11-02:
Hi @derekstraka now what about dunfell? :)

aj-bagwell, 2021-08-06:
The new override syntax (bf6705fb7f41b16c3277cbf84af0b9757dcd3a01) breaks rust on the old release, would it be possible to create branches for them?

srwalter, 2021-08-30:
Which release are you interested in?

aj-bagwell, 2021-09-07:
`hardknott` and `dunfell` (LTS) at a minimum as they are the two releases supported. But I would not mind `zeus` and `gatesgarth` too.

srwalter, 2021-09-19:
Created hardknott, zeus, and gatesgarth branches

closed 2021-09-19 by srwalter