meta-rust/meta-rust #220: build cargo-native fails on sumo (libssh2-sys)

stfl, 2018-12-05:
## Version(s) of meta-rust

sumo

## Version(s) of poky and/or oe-core

sumo

## Actual result

```
error[E0425]: cannot find function `read_to_string` in module `fs`
  --> /yocto/weinmann/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:99:26
   |
99 |         let config = fs::read_to_string("libssh2/src/libssh2_config_cmake.h.in")
   |                          ^^^^^^^^^^^^^^ did you mean `read_string`?

error[E0425]: cannot find function `read_to_string` in module `fs`
   --> /yocto/weinmann/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:124:24
    |
124 |     let libssh2h = fs::read_to_string("libssh2/include/libssh2.h").unwrap();
    |                        ^^^^^^^^^^^^^^ did you mean `read_string`?

error[E0425]: cannot find function `read_to_string` in module `fs`
   --> /yocto/weinmann/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:134:13
    |
134 |         fs::read_to_string("libssh2/libssh2.pc.in")
    |             ^^^^^^^^^^^^^^ did you mean `read_string`?

error: use of unstable library feature 'fs_read_write' (see issue #46588)
   --> /yocto/weinmann/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/libssh2-sys-0.2.11/build.rs:105:9
    |
105 |         fs::write(build.join("libssh2_config.h"), &config).unwrap();
    |         ^^^^^^^^^

error: use of unstable library feature 'fs_read_write' (see issue #46588)
```

## Steps to reproduce

build from scratch on sumo

apperently read_to_string is only available in rust >1.26 and

siredmar, 2019-03-05:
I have the same issue building on sumo with cargo-native 0.25

agherzan, 2019-05-01:
I have the same issue and I'm really curious how could this happen given that the create versions are fixed in the recipe.