meta-rust/meta-rust #224: error: no matching package named `atty` found (required by `cargo`)

paroque28, 2018-12-19:
## Version(s) of meta-rust
sumo branch
## Version(s) of poky and/or oe-core
 DISTRO_VERSION = "2.5.1"
 DISTRO_CODENAME = "sumo"
## Expected result
Build completed
## Actual result
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: cargo-native-0.25.0-r0 do_compile: Function failed: do_compile (log file is located at /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.141829)
ERROR: Logfile of failure stored in: /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.141829
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo
| NOTE: rustc =
| NOTE: /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo build -v --target x86_64-linux --release
| error: no matching package named `atty` found (required by `cargo`)
| location searched: registry `https://github.com/rust-lang/crates.io-index`
| version required: = 0.2.11
| WARNING: /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/run.do_compile.141829:1 exit 101 from '"/wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo" build -v --target x86_64-linux --release "$@"'
| ERROR: Function failed: do_compile (log file is located at /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.141829)
ERROR: Task (virtual:native:/wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/../layers/meta-rust/recipes-devtools/cargo/cargo_0.25.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3962 tasks of which 3960 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/../layers/meta-rust/recipes-devtools/cargo/cargo_0.25.0.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
[000000029][LOG]Build for edison failed. Check failed log in build/tmp/log/cooker/edison .
[000000029][LOG]If build for edison succeeded, final image should have been generated here:
[000000029][LOG]   build/tmp/deploy/images/edison/resin-image-edison.zip
[000000029][LOG]Done.
## Steps to reproduce
$ bitbake image

Do you have any ideas? I am trying to build balenaOS

paroque28, 2018-12-19:
balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.141829:

DEBUG: Executing shell function do_compile
 NOTE: cargo = /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo
 NOTE: rustc =
 NOTE: /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo build -v --target x86_64-linux --release
 error: no matching package named `atty` found (required by `cargo`)
 location searched: registry `https://github.com/rust-lang/crates.io-index`
 version required: = 0.2.11
 WARNING: /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/run.do_compile.141829:1 exit 101 from '"/wr/sandbox/prodrigu/cases/balenaOS/balena-edison/  build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo" build -v --target x86_64-linux --release "$@"'
 ERROR: Function failed: do_compile (log file is located at /wr/sandbox/prodrigu/cases/balenaOS/balena-edison/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.141829)
 ~
 ~
 ~
 ~
 ~
 ~
 ~
 ~
 ~
 ~
 ~
 ~
 ~

siredmar, 2019-03-04:
I have the exact same problem here building this. Also trying to build balenaOs

siredmar, 2019-03-07:
I found out it is somehow related to this commit: https://github.com/meta-rust/meta-rust/commit/48e2a7db2588d55ee5c43cf5242ad41aa16a2c1c#diff-86fe95c88946300256cccb34400231e7

The crates server gets configured to something "/nonexistant" resulting in cargo not able to install any crate.
Other people don't have that kind of problem with the same repository. Question is: how and why does this gets configured in the first place? If i get rid of those lines
```
[source.crates-io]
		replace-with = "bitbake"
		local-registry = "/nonexistant"
```
cargo-native can pull the crates properly.

ColinFinck, 2019-03-07:
I fixed building our Rust application with meta-rust by adding `CARGO_DISABLE_BITBAKE_VENDORING = "1"` to our recipe. This reenables Cargo to fetch dependencies on its own.
The only other alternative seems to be using cargo-bitbake to create a recipe with all dependencies explicitly mentioned.

srwalter, 2019-03-07:
@ColinFinck would you mind opening in issue in https://github.com/meta-rust/meta-rust/ explaining why cargo-bitbake doesn't work for you?  That is indeed the intended way to use the cargo.bbclass

bernardoaraujor, 2021-01-22:
any progress on this issue?
I'm having similar problems. Adding `CARGO_DISABLE_BITBAKE_VENDORING = "1"` to my recipe takes me to error messages similar to reported [here](https://github.com/meta-rust/cargo-bitbake/issues/15).

bhass1, 2022-01-06:
I faced a similar issue using branch `gatesgarth` while building [`watchexec`](https://github.com/watchexec/watchexec). 
```
DISTRO_VERSION="3.2"; 
DISTRO_CODENAME=gatesgarth
```

Error output
```
ERROR: watchexec-cli-1.17.1-r0 do_compile: Execution of '/REDACTED/watchexec-cli/1.17.1-r0/temp/run.do_compile.1442111' failed with exit code 101:
error: no matching package named `clearscreen` found
location searched: registry `https://github.com/rust-lang/crates.io-index`
required by package `watchexec v1.17.1 (/REDACTED/watchexec-cli/1.17.1-r0/git/lib)`
WARNING: exit code 101 from a shell command.

ERROR: Logfile of failure stored in: /REDACTED/watchexec-cli/1.17.1-r0/temp/log.do_compile.1442111
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /REDACTED/watchexec-cli/1.17.1-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target aarch64-fslc-linux --release --manifest-path=/REDACTED/watchexec-cli/1.17.1-r0/git//Cargo.toml --bin=watchexec
| error: no matching package named `clearscreen` found
| location searched: registry `https://github.com/rust-lang/crates.io-index`
| required by package `watchexec v1.17.1 (/REDACTED/watchexec-cli/1.17.1-r0/git/lib)`
| WARNING: exit code 101 from a shell command.
| ERROR: Execution of '/REDACTED/watchexec-cli/1.17.1-r0/temp/run.do_compile.1442111' failed with exit code 101:
| error: no matching package named `clearscreen` found
| location searched: registry `https://github.com/rust-lang/crates.io-index`
| required by package `watchexec v1.17.1 (/REDACTED/watchexec-cli/1.17.1-r0/git/lib)`
| WARNING: exit code 101 from a shell command.
|
ERROR: Task (/REDACTED/watchexec-cli_1.17.1.bb:do_compile) failed with exit code '1'
```

Adding `CARGO_DISABLE_BITBAKE_VENDORING = "1"` fixes the issue.

Does the meta-rust layer pin the version of the crates index it uses? Perhaps there's a version mismatch between meta-rust's crates index and the version of the crate required to compile?