meta-rust/meta-rust #240: TUNE_FEATURES parsing needs updating since warrior

ZubairLK, 2019-05-30:
## Version(s) of meta-rust
master https://github.com/meta-rust/meta-rust/commit/84710b341be53cc2af13ecc8ae87773cf84687bf
## Version(s) of poky and/or oe-core
Warrior branch https://github.com/openembedded/openembedded-core/tree/warrior
## Expected result
Compile libstd-rs successfully
## Actual result
Failed to compile libstd-rs with error: `(signal: 11, SIGSEGV: invalid memory reference)`

```
NOTE: cargo = /home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot-native/usr/bin/cargo
NOTE: rustc = 
NOTE: cargo build -vv --target arm-poky-linux-gnueabi --release --features 'panic-unwind backtrace' 
cargo 1.33.0
rustc 1.33.0
       Fresh cc v1.0.28
   Compiling core v0.0.0 (/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libcore)
     Running `CARGO=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot-native/usr/bin/cargo CARGO_PKG_VERSION_MINOR=0 CARGO_PKG_HOMEPAGE= CARGO_PKG_VERSION_PRE= CARGO_MANIFEST_DIR=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libcore LD_LIBRARY_PATH='/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps:/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot-native/usr/lib:/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/sysroots-components/x86_64/pseudo-native/usr/lib/pseudo/lib:/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/sysroots-components/x86_64/pseudo-native/usr/lib/pseudo/lib64' CARGO_PKG_DESCRIPTION= CARGO_PKG_REPOSITORY= CARGO_PKG_NAME=core CARGO_PKG_VERSION_MAJOR=0 CARGO_PKG_AUTHORS='The Rust Project Developers' CARGO_PKG_VERSION_PATCH=0 CARGO_PKG_VERSION=0.0.0 rustc --crate-name core src/libcore/lib.rs --color always --crate-type lib --emit=dep-info,link -C opt-level=2 -C metadata=f3ee7b3c57c78f00 -C extra-filename=-f3ee7b3c57c78f00 --out-dir /home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/wrapper/target-rust-ccld -L dependency=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/arm-poky-linux-gnueabi/release/deps -L dependency=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps -L /home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0=/usr/src/debug/libstd-rs/1.33.0-r0 -L /home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot/usr/lib -C link-arg=-Wl,-soname,libstd.so`
   Compiling std v0.0.0 (/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd)
     Running `/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/build/std-71419368f3260db6/build-script-build`
[std 0.0.0] cargo:rustc-link-lib=dl
[std 0.0.0] cargo:rustc-link-lib=rt
[std 0.0.0] cargo:rustc-link-lib=pthread
error: Could not compile `core`.

Caused by:
  process didn't exit successfully: `CARGO=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot-native/usr/bin/cargo CARGO_PKG_VERSION_MINOR=0 CARGO_PKG_HOMEPAGE= CARGO_PKG_VERSION_PRE= CARGO_MANIFEST_DIR=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libcore LD_LIBRARY_PATH='/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps:/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot-native/usr/lib:/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/sysroots-components/x86_64/pseudo-native/usr/lib/pseudo/lib:/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/sysroots-components/x86_64/pseudo-native/usr/lib/pseudo/lib64' CARGO_PKG_DESCRIPTION= CARGO_PKG_REPOSITORY= CARGO_PKG_NAME=core CARGO_PKG_VERSION_MAJOR=0 CARGO_PKG_AUTHORS='The Rust Project Developers' CARGO_PKG_VERSION_PATCH=0 CARGO_PKG_VERSION=0.0.0 rustc --crate-name core src/libcore/lib.rs --color always --crate-type lib --emit=dep-info,link -C opt-level=2 -C metadata=f3ee7b3c57c78f00 -C extra-filename=-f3ee7b3c57c78f00 --out-dir /home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/wrapper/target-rust-ccld -L dependency=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/arm-poky-linux-gnueabi/release/deps -L dependency=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps -L /home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0=/usr/src/debug/libstd-rs/1.33.0-r0 -L /home/zubairlk/resin/yocto/balena-raspberrypi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot/usr/lib -C link-arg=-Wl,-soname,libstd.so` (signal: 11, SIGSEGV: invalid memory reference)
WARNING: exit code 101 from a shell command.
```

## Steps to reproduce

Compile balenaOS for the raspberrypi3 device using instructions here
https://www.balena.io/os/docs/custom-build/
but use warrior PR for the meta-balena layer,
https://github.com/balena-os/meta-balena/pull/1520
and switch branches of poky/meta-oe layers to warrior.

I've traced the issue to this OE commit https://github.com/openembedded/openembedded-core/commit/ac83d22eb5031f7fdd09d34a1a46d92fd3e39a3c by @kraj 

This removes `armv7ve` from `TUNE_FEATURES` which is parsed in meta-rust in rust.inc.

https://github.com/meta-rust/meta-rust/blob/84710b341be53cc2af13ecc8ae87773cf84687bf/recipes-devtools/rust/rust.inc#L72-L85

I managed to get libstd to compile by using the following temporary workaround.
```
+++ /home/zubairlk/resin/yocto/balena-raspberrypi/layers/meta-rust/recipes-devtools/rust/rust.inc
@@ -88,6 +88,10 @@
         f.append("+a5")
     if 'cortexa7' in feat:
         f.append("+a7")
+        if "+v7" not in f:
+            f.append("+v7")
+        if "+thumb2" not in f:
+            f.append("+thumb2")
     if 'cortexa9' in feat:
         f.append("+a9")
     if 'cortexa15' in feat:
```

I'm guessing this will affect other arm devices as well as the OE patch changes TUNE_FEATURES for all arches. So the parsing in `rust.inc` will need updating. Open to suggestions for a generic way. If the above is ok, I can submit a PR for it.

ZubairLK, 2019-06-03:
I can easily reproduce the error using poky using the following commands

```
git clone git://git.yoctoproject.org/poky
cd poky/
git clone git://git.openembedded.org/meta-openembedded 
git clone https://github.com/meta-rust/meta-rust.git
git clone git://git.yoctoproject.org/meta-raspberrypi
source oe-init-build-env 
bitbake-layers add-layer ../meta-openembedded/meta-oe
bitbake-layers add-layer ../meta-raspberrypi
bitbake-layers add-layer ../meta-rust/
MACHINE=raspberrypi3 bitbake libstd-rs
```

```
|      Running `rustc --crate-name build_script_build /home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/vendor/bac
ktrace-sys/build.rs --color never --crate-type bin --emit=dep-info,link -C opt-level=2 --cfg 'feature="compiler_builtins"' --cfg 'feature="core"' --cfg 'feature="rustc-dep-of-std"' -C m
etadata=788ee2e2242e9ca4 -C extra-filename=-788ee2e2242e9ca4 --out-dir /home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33
.0-src/src/libstd/release/build/backtrace-sys-788ee2e2242e9ca4 -C linker=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/wrapper/
build-rust-ccld -L dependency=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps --extern c
c=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps/libcc-ccd85df0d100aae8.rlib --cap-lint
s allow`
|      Running `rustc --crate-name build_script_build src/libstd/build.rs --color never --crate-type bin --emit=dep-info,link -C opt-level=2 --cfg 'feature="backtrace"' --cfg 'feature="
backtrace-sys"' --cfg 'feature="compiler_builtins"' --cfg 'feature="compiler_builtins_c"' --cfg 'feature="default"' --cfg 'feature="panic-unwind"' --cfg 'feature="panic_unwind"' -C meta
data=71419368f3260db6 -C extra-filename=-71419368f3260db6 --out-dir /home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-
src/src/libstd/release/build/std-71419368f3260db6 -C linker=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/wrapper/build-rust-cc
ld -L dependency=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps --extern cc=/home/Zubai
rLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps/libcc-ccd85df0d100aae8.rlib`
|      Running `/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/build/compiler_builtins-28318
221649df97c/build-script-build`
|      Running `/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/build/unwind-b04ef420ba73343a
/build-script-build`
|      Running `/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/build/backtrace-sys-788ee2e22
42e9ca4/build-script-build`
|      Running `/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/build/std-71419368f3260db6/bu
ild-script-build`
| error: Could not compile `core`.
| 
| Caused by:
|   process didn't exit successfully: `rustc --crate-name core src/libcore/lib.rs --color never --crate-type lib --emit=dep-info,link -C opt-level=2 -C metadata=d2f3ce5d5651732b -C extr
a-filename=-d2f3ce5d5651732b --out-dir /home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/arm-poky-linux
-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/wrapper/target-ru
st-ccld -L dependency=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/arm-poky-linux-gnueabi/release/
deps -L dependency=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/rustc-1.33.0-src/src/libstd/release/deps -L /home/ZubairLK/res
in/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2
hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0=/usr/src/debug/libstd-rs/1.33.0-r0 -L /home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/
1.33.0-r0/recipe-sysroot/usr/lib -C link-arg=-Wl,-soname,libstd.so` (signal: 11, SIGSEGV: invalid memory reference)
| WARNING: exit code 101 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/ZubairLK/resin/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.33.0-r0/temp/log.do_compile.
3632)
No currently running tasks (585 of 592)  98% |##############################################ERROR: Task (/home/ZubairLK/resin/poky/meta-rust/recipes-devtools/rust/libstd-rs_1.33.0.bb:do
_compile) failed with exit code '1'
No currently running tasks (585 of 592)  98% |##############################################NOTE: Tasks Summary: Attempted 585 tasks of which 0 didn't need to be rerun and 1 failed.
No currently running tasks (585 of 592)  98% |##############################################
Summary: 1 task failed:
  /home/ZubairLK/resin/poky/meta-rust/recipes-devtools/rust/libstd-rs_1.33.0.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

floion, 2019-10-29:
Hi, any updates for this issue?

closed 2019-12-15 by srwalter

floion, 2020-11-02:
@srwalter what was the resolution for this?

PR #242 (2019-12-15): rust: Improve TUNE_FEATURE parsing
Since https://github.com/openembedded/openembedded-core/commit/ac83d22eb5031f7fdd09d34a1a46d92fd3e39a3c

The armvX arch definition is not present in TUNE_FEATURES but is now in
MACHINEOVERRIDES. Improve parsing so that MACHINEOVERRIDES is also checked.

Fixes #240

Tested on raspberrypi3 with balenaOS which uses rust.
commit 186ec590 (2019-12-15): rust: Improve TUNE_FEATURE parsing

Since https://github.com/openembedded/openembedded-core/commit/ac83d22eb5031f7fdd09d34a1a46d92fd3e39a3c

The armvX arch definition is not present in TUNE_FEATURES but is now in
MACHINEOVERRIDES. Improve parsing so that MACHINEOVERRIDES is also checked.

Fixes #240

Tested on raspberrypi3 with balenaOS which uses rust.